VLSIDA/OpenRAM #9: vlsilayout.py is not able to catch ports/labels from gds file created by glade

anujnagar, 2016-11-24:
Hello mguthaus,

While am running 04_pinv_test.py for  tsmc technology, i am getting an error like 
"ERROR: runTest (__main__.pinv_test)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/anagar/Templates/rcmc/OpenRAM-master/compiler/tests/04_pinv_test.py", line 24, in runTest
    import pinv
  File "/home/anagar/Templates/rcmc/OpenRAM-master/compiler/tests/../pinv.py", line 10, in <module>
    class pinv(design.design):
  File "/home/anagar/Templates/rcmc/OpenRAM-master/compiler/tests/../pinv.py", line 16, in pinv
    c = reload(__import__(OPTS.config.bitcell))
  File "/home/anagar/Templates/rcmc/OpenRAM-master/compiler/tests/../bitcell.py", line 6, in <module>
    class bitcell(design.design):
  File "/home/anagar/Templates/rcmc/OpenRAM-master/compiler/tests/../bitcell.py", line 15, in bitcell
    chars = utils.auto_measure_libcell(pins, "cell_6t", GDS["unit"], layer["boundary"])
  File "/home/anagar/Templates/rcmc/OpenRAM-master/compiler/tests/../utils.py", line 42, in auto_measure_libcell
    cell[str(pin)] = gdsPinToOffset(cell_vlsi.readPin(str(pin)))
  File "/home/anagar/Templates/rcmc/OpenRAM-master/compiler/gdsMill/gdsMill/vlsiLayout.py", line 673, in readPin
    pin_boundary=[pin_boundary[0]*self.units[0],pin_boundary[1]*self.units[0],pin_boundary[2]*self.units[0],pin_boundary[3]*self.units[0]]
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'

----------------------------------------------------------------------
Ran 1 test in 0.085s

FAILED (errors=1)"

Can you please have a look on it, what i am missing here.

Thanks,
Anuj Nagar

mguthaus, 2016-11-24:
Hi,

Which technology are you running? Can you provide more information so that
we can replicate this?

mguthaus, 2016-11-24:
Also,

Please make sure you are using the most recent version. Can you send us the
output is git log?

anujnagar, 2016-11-28:
TSMC technology i am using.
Where i can get this git log output?

I was running the test case for 04_pinv_test.py. It was asking for bitcell gds file thn i copied gds file to gds_lib dir and again i ran it. After that it gave me the above error.

mguthaus, 2016-11-28:
Hi Anuj,

The 04_pinv_test.py requires a bitcell to be defined and passing in order
to match the bitcell height. You should get the tests passing in order by
starting with test 00, 01, etc. Does test 01 pass? Though, this doesn't
attempt to measure read the pins of the bitcell, so you may not see the
error with that cell until test 04.

The bitcell likely doesn't pass if you did not create the correct labels
for all pins. The pin names are defined in the technology file. I agree
that this error message is cryptic and we will add a better one.

Matt

anujnagar, 2016-11-28:
Does port name should match with freepdk45 bitcell ?
Anyhow if here we are using bit cell for height only thn port matching doesn't make sense.
Okay i will check it again and i will let you know.

mguthaus, 2016-11-28:
There are example names provided in the freepdk45 bitcell. You can change
the names in the technology file. There may be places that these are
hard-coded so it is best to follow the same names. However, we would like
to ensure they aren't hard coded in the future.

We need to ensure that we can make an inverter that matches the pitch of
the bitcell. If not, this test should fail because we won't be able to make
an SRAM.

anujnagar, 2016-11-28:
one more thing, how can i check the code version.
I mean how can i check  which version i am using.

mguthaus, 2016-11-28:
If you cloned from github, you should use git by typing:

git log

and see the most recent commit hash. Right now, the master (released copy)
is at:

commit d195df682d1788e8f3b9a9e9d9f693c6e5d8c76d

anujnagar, 2016-11-28:
Hello Matt,
      
   Is there any way to generate spuce and verilog only.
If we don't have layout of basic leaf cells. Cant we get spice and verilog for simple circuit verification.

mguthaus, 2016-11-28:
Sorry:

commit 2d0533a7d5befaded2d84d3f5109c3834fe9c122

I was on a different branch.

mguthaus, 2016-11-28:
Hi Anuj,

That is not currently an option, but we could add it as a feature request.

closed 2016-11-28 by mguthaus