VLSIDA/OpenRAM #10: gds2writer.py is not able to generate gds file

anujnagar, 2016-12-06:
Hello Matt,

I am facing an issue, while i am writing a gds file then gds2writer.py is showing an error like 
" writer.writeToFile("sqrin.gds")
  File "/home/anagar/work/cwork/pycells/gdsmillPkg/gdsMill/gds2writer.py", line 539, in writeToFile
    self.writeGds2()
  File "/home/anagar/work/cwork/pycells/gdsmillPkg/gdsMill/gds2writer.py", line 529, in writeGds2
    self.writeHeader();  #first, put the header in
  File "/home/anagar/work/cwork/pycells/gdsmillPkg/gdsMill/gds2writer.py", line 160, in writeHeader
    dbUnits=self.ibmDataFromIeeeDouble((self.layoutObject.info["units"][0]*1e-6/self.layoutObject.info["units"][1])*self.layoutObject.info["units"][1])
TypeError: unsupported operand type(s) for /: 'float' and 'instancemethod' "

I check it with many files like below :- 
1. I read a gds file (by VlsiLayout) generated by layout editor tool.
2. I made some changes and generates its  gds file
3. Again i read the gds file generated by gds2writer 
 4. i  made some changes and while i am trying to generate gds file by gds2writer that time i am facing 
error.

So i conclude that gds2writer is not able to generate the file if we are reading the gds file generated by gds2writer.

Matt, Can you please help me 

Thanks,
Anuj Nagar

mguthaus, 2016-12-06:
Hi Anuj,

Since this isn't really an issue with openram itself and I cannot replicate
it with a test case, I'm not sure how to help...

Matt

mguthaus, 2016-12-06:
I'm closing for now, but if you can provide a test case it would be useful.

closed 2016-12-06 by mguthaus

anujnagar, 2016-12-06:
Hello Matt,
 I think i was not able to explain you. what problem exactly i am facing.