VLSIDA/OpenRAM #28: Totally, Unit test error with tech: freepdk45 pycell

leepil, 2017-08-17:
Hello mguthaus.  
I am  at the beginning of most entry. 
but  when i responded Readme, Unit test clogged up in this part. 

For info, 00_code_format_check_test.py is pass,  I installed freepdk45 tech. All tool's version is correct : python 2.7.9 also h,ng spice . 

01~30(openram total test) are failed. because,,

ERROR: runTest (30_openram_test.openram_test)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/leepil/OpenRAM-master/compiler/tests/30_openram_test.py", line 23, in runTest
    globals.init_openram("config_20_{0}".format(OPTS.tech_name))
  File "/home/leepil/OpenRAM-master/compiler/tests/../globals.py", line 113, in init_openram
    import_tech()
  File "/home/leepil/OpenRAM-master/compiler/tests/../globals.py", line 314, in import_tech
    __import__(filename)
  File "/home/leepil/OpenRAM-master/technology/setup_scripts/setup_openram_freepdk45.py", line 16, in <module>
    PDK_DIR=os.path.abspath(os.environ.get("FREEPDK45"))
  File "/home/leepil/PycellStudio3/plat_linux_gcc44x_64/3rd/lib/python2.7/posixpath.py", line 367, in abspath
    if not isabs(path):
  File "/home/leepil/PycellStudio3/plat_linux_gcc44x_64/3rd/lib/python2.7/posixpath.py", line 61, in isabs
    return s.startswith('/')

AttributeError: 'NoneType' object has no attribute 'startswith'

This error appeared as common between 01 and 30. 

can you help me please?

Thank you

closed 2017-08-17 by leepil

mguthaus, 2017-08-17:
Hi,

It seems that maybe you have not defined the FREEPDK45 directory to point
to the PDK directory? Can you send what it is defined as in your
environment?

echo $FREEPDK45
echo $OPENRAM_TECH
echo $OPENRAM_HOME

Matt

leepil, 2017-08-22:
Thank you for the reply.

In setpaths.csh...
    $setenv OPENRAM_HOME /home/leepil/OpenRAM-master/compiler
    $setenv OPENRAM_TECH /home/leepil/OpenRAM-master/technology
    $setenv FREEPDK45 /home/leepil/OpenRAM-master/technology/freepdk45

You used `pwd`, but I set the path directly.

mguthaus, 2017-08-22:
Did you source this file in your shell? Are the variables actually set?
Please run the commands that I suggested to confirm.

leepil, 2017-08-23:
That's true! You taght me that " I will not change the code given.". Thx
but... A new error occurred. I obviously made 'calibre' set correctly(of course, i am accurate within knowledge i know).

Traceback (most recent call last):
  File "/home/leepil/OpenRAM-master/compiler/tests/20_sram_1bank_test.py", line 31, in runTest
    self.local_check(a)
  File "/home/leepil/OpenRAM-master/compiler/tests/20_sram_1bank_test.py", line 42, in local_check
    self.assertFalse(calibre.run_drc(a.name, tempgds))
  File "/home/leepil/OpenRAM-master/compiler/tests/../calibre.py", line 116, in run_drc
    debug.error("Unable to retrieve DRC results file. Is calibre set up?",1)
  File "/home/leepil/OpenRAM-master/compiler/tests/../debug.py", line 24, in error
    sys.exit(return_value)
SystemExit: 1

This error is commonly shown for about 10 tests.

p.s
(I tried to solve openram in the course of two months after I wanted to run openram, but I would like to inform you that it was not done. I also saw the caliber related error I showed as I had already seen. Finally, I ask for help if I can not solve it at all. Thank you very much.)

mguthaus, 2017-08-23:
Check the OpenRAM temp directory for the output from Calibre. This is
discussed in the presentation for how to debug. This is likely a problem
running (or getting a license for ( Calibre.