VLSIDA/OpenRAM #41: Feature: dual port SAM (sequential access memory)

jeras, 2018-07-26:
In many DSP applications (stream processing) memories are only accessed sequentially, so instead of using an address decoder a one hot shift register could be used. Dual ports should be available (one write, one read), since reads are generally not performed from the same address as writes, otherwise such a memory would only offer a fixed delay.

I could use such memories in a project a few years ago. Images were processed in a stream, each memory contained a few lines of the image. A single line was written while simultaneously multiple lines were read out and used in a processing kernel (demosaicing, 2D FIR filters, resizing, ...).

In many such applications memory contents are always overwritten very soon, for example in image processing the maximum time the memory would need to retain data would be at most the time between two frames (42ms at 24fps) and at least the time between two lines (in the range of 10us).
This means dynamic memory cells could be used, without redundant refresh logic (present in all commercial 1T memories). Additionally many such applications can tolerate small error rates, so it should be possible to use very small dynamic cells, thus reducing area and power.

I mentioned image processing since memory requirements can be rather high, and since I have experience with it. The same principles can be applied to audio and radio signals. Since many AI systems are area hungry and noise tolerant, the same principles might be applicable there too, and this is a rapidly growing market.

Right now there are only few SAM research articles available and at the time I could find no commercial memory compilers. And I never saw mentioned a combination of SAM and low retention time dynamic memory cells.

mguthaus, 2018-07-26:
Thanks for the feature request. This is a great suggestion and I appreciate you taking the time to post this. We will keep this open while we can get to it. 

We are in the process of a major rewrite to fix some power supply organization and other issues that will let us scale to newer technologies (7nm, 14nm) and multiport more easily. These are taking priority for now.

jeras, 2018-07-27:
The SAM article I was referring to is from 2002:
Low-Power Sequential Access Memory Design
Joong-Seok Moon, William C. Athas, Peter A. Beerel, Jeffrey T. Draper
https://www.isi.edu/~draper/papers/cicc02.pdf

Apparently there is a patent from 2004 owned by Aptina (now ON semi), strangely I did not know about it. I will not provide links and there obviously is some prior art.

In a SAM, even if replacing the address controller with a one hot shift register does not save much area, there is no need to wire the address bus from a state machine to the memory, this should improve timing. Address counters are generally still needed, to prevent underflow/overflow, but there is no need to connect them to the memory.

The other topic was pseudo static RAM. This is DRAM with a SRAM like interface, a controller handling refresh and similar hides most of the overhead.
https://en.wikipedia.org/wiki/1T-SRAM
https://en.wikipedia.org/wiki/Dynamic_random-access_memory#PSRAM
As I mentioned there are use cases where DRAM data retention is not an issue, since data is overwritten at a higher rate then the refresh cycle. While some dedicated DRAM circuitry might still be needed (I am not an expert), it should be possible to avoid the area overhead of a refresh circuit and SRAM cache. While searching a few years ago I was unable to find a PSRAM without controller overhead. The overhead might not seem much for a single large memory (CPU data), but is significant if many small independent memories are needed (FIR, FIFO, ... buffers in a complex pipeline).

mguthaus, 2022-07-21:
We would accept this as a contribution but it is not on the roadmap.

closed 2022-07-21 by mguthaus