VLSIDA/OpenRAM #42: CPU memory with unaligned access

jeras, 2018-07-31:
I will just describe an example. The code for RISC-V CPU with the compressed instruction extension is a mix of 32 and 16 bit instructions, this meas 32bit instructions are not always 32bit aligned.

If the CPU requests an unaligned 32bit instruction, two memory read cycles must be performed to fetch it. A buffer containing the previous read can be used to avoid repeating the read, but this it does not help if a branch is taken.

A RAM/ROM with a 32bit data bus and addressing granularity of 16bit would enable CPU instruction fetches without overhead. There are use cases where reading only 16bit data would be needed (if in the previous cycle 32bits were read containing a 16bit instruction and half of a 32bit instruction).

The same functionality can be achieved with a pair of memories with 16bit data buses and some extra logic. But a dedicated solution could probably offer better power optimizations.

Open questions are:
* how to handle the end of the array (would it wrap),
* how to combine multiple memories,
* little/big endian,
* are there enough area and speed improvements to make it worth?

mguthaus, 2018-08-06:
This seems more like a RISC-V implementation question. You should consider
asking someone in that area.

---
Matthew Guthaus
Professor, Computer Engineering
University of California Santa Cruz
https://www.soe.ucsc.edu/people/mrg

jeras, 2018-08-07:
Well, this is a request for memory generator functionality.

But you are right, the usefulness of such approach should be discussed by users (RISC-V devs) first.
https://groups.google.com/a/groups.riscv.org/forum/#!topic/hw-dev/MF7AWzzztpM

mguthaus, 2018-08-08:
Memories do not directly deal with aligned and unaligned addresses -- there 
is no such thing from a memory point of view. All memory interfaces will 
have an address and read or write the data as appropriate. If you want to 
have a 32-bit address and select half of the data word, it is up to the 
state machine of the logic that does the fetching to select half of the 
data. If this requires a second fetch for the other half, the state machine 
should be able to do that as well.

Hope that helps,
Matt

closed 2018-08-14 by mguthaus