VLSIDA/OpenRAM #43: Logical Error in Hierarchical Decoder

lekez2005, 2018-11-29:
The hierarchical decoder is incorrect in terms of mapping an input address to the correct row. For example, for a 64 row decoder, address 1 mapped to row 16 and address 32 mapped to row 2. Unfortunately, my current decoder has widely diverged from the existing implementation so I can't generate a good enough clean patch.
 [This gist](https://gist.github.com/lekez2005/2a1f61a6fd873fd4d14a2c744a90c7a3) shows my patch for fixing the issue. At the very least, whatever fix for this issue should pass the included tests. [This gist](https://gist.github.com/lekez2005/2b40bd5445e0e735a3c80a7f7c94b083) shows the full decoder for reference.

mguthaus, 2018-11-29:
Hi,

Yes, this is fixed in our internal copy and will be released soon.

Thanks for letting us know.

Matt

mguthaus, 2018-11-29:
Resolved in upcoming release.

closed 2018-11-29 by mguthaus