VLSIDA/OpenRAM #44: Verilog vs liberty output mismatch

tetra12, 2018-12-12:
Issue: Bug
Severity: High
Description: When I generate a default SRAM from your example for the latest gscl45 with -n option, the output **verilog** has **OEb pin**, which is NOT available in the **liberty** output. This results in a synthesis fail when I synthesize a wrapper for the generated memory. Pls fix this

mguthaus, 2018-12-12:
Thanks, we are aware of this issue and it will be fixed in an upcoming release very soon.

We have separated the data bus into separate din and dout ports so the OEb is no longer necessary but was not removed from the verilog.

tetra12, 2018-12-12:
Excellent, thank you! So I'll simply remove this **OEb** port from my verilog.

Another issue is related. Yosys fails to parse the liberty output in current format:
    pin(DOUT0[63:0]){ ... }

My workaround is to remove brackets [63:0] and leave pin (DOUT0) {...}. Please remove this from your scripts as well

tetra12, 2018-12-16:
Thank you. Waiting forward working with your next release. Any contributions required ?

tetra12, 2018-12-16:
@mguthaus Yosys parses the *.lib files if you remove bus widths from pin descriptions :  
pin(DOUT0[63:0]){ ... } -> pin(DOUT0){ ... }

I also made a verilog wrapper for your memories to match lib files. I was able to synthesize it with an updated lib file in Yosys.

mguthaus, 2018-12-16:
But won't that break most other static timers (Primetime, OpenSTA)? They
need to recognize each bus pin...

mguthaus, 2018-12-16:
Maybe this is a bug in how yosys parses .lib files...

---
Matthew Guthaus
Professor, Computer Science and Engineering
University of California Santa Cruz
https://www.soe.ucsc.edu/people/mrg

tetra12, 2018-12-16:
Pls leave your comment here too: https://www.reddit.com/r/yosys/comments/a4ovm6/liberty_syntax_support/

I'm working with Tim now. No problems with qflow with that. I don't have access to commercial tools, so pls check on your own

mguthaus, 2018-12-17:
> Did you confirm the timing was right or that it just read it in?
> I confirm that updated *.lib file works seamlessly with opensta.
> Synthesized, checked.
> Please, follow my feedback and remove bus width in the next release.
> Here's my wrapper
> your SRAM. Feel free to update your verilog output as well

My guess is that the will make adding timing constraints problematic. These
need to use indexed buses like DOUT0[1] so removing the indices may prevent
that. Even if it does "work" does it then count all of the bits as the same
timing point? That would not be correct either.

I think we should instead fix Yosys.

tetra12, 2018-12-17:
I synthesized your SRAM with my verilog wrapper and run the opensta script to obtain timing report. I used OSU035 with a 10ns clock, which yielded in a 9.86ns slack. Do you want to get my scripts ?

tetra12, 2018-12-17:
If you think it is the Yosys issue, pls feel free to open a RFC for this in the Yosys repo. If you want to double check my scripts, let me know. I'll share

mguthaus, 2018-12-17:
Yes, please share your scripts.

---
Matthew Guthaus
Professor, Computer Science and Engineering
University of California Santa Cruz
https://www.soe.ucsc.edu/people/mrg

tetra12, 2018-12-18:
Any follow up questions? Should I close this issue ?

mguthaus, 2018-12-18:
I'm on vacation right now, so no... We will look at this for the next
release.

---
Matthew Guthaus
Professor, Computer Science and Engineering
University of California Santa Cruz
https://www.soe.ucsc.edu/people/mrg

tetra12, 2019-01-04:
Happy New Year! Any news?

mguthaus, 2019-01-04:
Not yet, unfortunately class prep must take priority.

mguthaus, 2019-01-09:
I've confirmed that your suggested change to the lib file works in all
timing analyzers that I can use. So, I have added it to our private
repository and it will be in the next OpenRAM release -- probably by the
end of this week. Thanks for your patience.

tetra12, 2019-01-10:
Sounds great. Let me know if I can contribute in this project. I can do testing, validation, integration and verification, since I use your stuff in my commercial project.

If you have an idea, let me know

mguthaus, 2019-01-10:
Can you send me an email outside of GitHub at mrg@ucsc.edu. I'm interested
in learning more and seeing how we can collaborate.

Matt

mguthaus, 2019-02-22:
I finally got to trying this out in Library Compiler and it fails without these bus notations. 

```
Reading '/home/mrg/OpenRAMLibrary/deliverables/freepdk45/2019-02-21/sram_8b_512_1rw_freepdk45/sram_8b_512_1rw_freepdk45_TT_1p0V_25C.lib' ...
Error: Line 96, Cell 'sram_8b_512_1rw_freepdk45', bus 'DIN0', Invalid bus syntax is detected in 'DIN0'. (LBDB-176)
Error: Line 135, Cell 'sram_8b_512_1rw_freepdk45', bus 'DOUT0', Invalid bus syntax is detected in 'DOUT0'. (LBDB-176)
Error: Line 169, Cell 'sram_8b_512_1rw_freepdk45', bus 'ADDR0', Invalid bus syntax is detected in 'ADDR0'. (LBDB-176)
```
Unfortunately, I will need to add them back and we should fix Yosys.

mguthaus, 2019-02-23:
This issue was coincidentally opened 4 hours ago for Yosys:
https://github.com/YosysHQ/yosys/issues/825

closed 2019-08-29 by mguthaus