VLSIDA/OpenRAM #45: Pins are blocked for P&R in generated LEF file

Fatsie, 2019-05-23:
I generated a 1024x8 SRAM block with the scn4m_subm technology. When looking at the LEF files the (input) pins don't seem to be accessible for P&R.
The flip-flops have an OBS on metal1 and metal2 and on metal3 and metal4 the GND/VDD grid is generated.
My design was just test chip with an OpenRAM block, I circumvented the problem by manual connecting the pins out of the block.

mguthaus, 2019-05-23:
Please provide which git version you are using and a configuration file.

The public repo is a bit outdated so some of these may have been fixed
already.

---
Matthew Guthaus
Professor, Computer Science and Engineering
University of California Santa Cruz
https://www.soe.ucsc.edu/people/mrg

Fatsie, 2019-05-23:
I am using head: 76e2ab8
config:
```python
word_size = 8
num_words = 1024

tech_name = "scn4m_subm"
process_corners = ["TT"]
supply_voltages = [ 3.3 ]
temperatures = [ 25 ]

output_path = output_name = "SRAM{}x{}".format(num_words, word_size)

route_supplies = True

drc_name = "magic"
lvs_name = "netgen"
pex_name = "magic"
```

mguthaus, 2019-05-31:
Sorry for the delay. We are aware and see this problem. There are two fixes that will be available soon to resolve this:
1) The pins will be brought to M3 on the perimeter. 
2) There is a mistake in generating blockages of the leaf cells (6T, sense amp, DFF). It simply blocks all metal1 and metal2, instead of the detailed blockages in the leaf GDS.

mguthaus, 2019-07-05:
This is fixed in an upcoming release. Instead of blocking all metal1 and metal2 in the leaf cells, we provide detailed blockages instead.

closed 2019-07-05 by mguthaus