VLSIDA/OpenRAM #46: Antenna violation for wordline connected to GND in replica bitline

Fatsie, 2019-05-23:
I generated 1024x8 block with scn4m_subm technology. I converted it to tape-out on TSMC 0.35um with some simple layer operations.
I got antenna violations in the replica bitcell. There wordlines of the replica cells are connected directly to GND. As the connections are quite long it gives antenna violations. I solved it by manually adding diodes below the via that connects the wordlines to GND.
In smaller technlogies direct connections to VDD or GND will even be prohibited by the design rules. So I think the proper fix is to introduce a tie low cell for tieing the word lines to ground.

mguthaus, 2019-05-23:
Please provide which git version you are using and a configuration file.

---
Matthew Guthaus
Professor, Computer Science and Engineering
University of California Santa Cruz
https://www.soe.ucsc.edu/people/mrg

Fatsie, 2019-05-23:
I am using head: 76e2ab8
config:
```python
word_size = 8
num_words = 1024

tech_name = "scn4m_subm"
process_corners = ["TT"]
supply_voltages = [ 3.3 ]
temperatures = [ 25 ]

output_path = output_name = "SRAM{}x{}".format(num_words, word_size)

route_supplies = True

drc_name = "magic"
lvs_name = "netgen"
pex_name = "magic"
```

mguthaus, 2019-07-05:
This is fixed in an upcoming release where we move the replica bitline into the bitcell array. The wordlines are no longer tied to GND.

closed 2019-07-05 by mguthaus