VLSIDA/OpenRAM #47: Retain data missing in generated .lib for DOUT on rising edge of the clock

Fatsie, 2019-05-23:
The current SRAM design exposes the sense amplifier directly on the output. This means that during precharge the output of DOUT is in undetermined digital state. This is reflected in the verilog code in the following way:
```verilog
  always @(posedge clk0)
  begin
    ...
    DOUT0 = 8'bx;
```
This mean that if there is a path from DOUT to the input of a flip-flop that also is clocked on the rising edge, hold violations may occur.
I have to admit I am not a liberty file or STA expert but I don't seem to find the needed data in the .lib that allows to check for this hold violation. I would expect that retain data is needed for DOUT on rising edge of the clock to make this possible.
I am currently targeting 0.35um so with a well controlled skew on the clock tree hold violations should not be a problem but for smaller nodes I think this becomes a necessity.

mguthaus, 2019-05-23:
This would be in the hold data of the receiving flip flop.

---
Matthew Guthaus
Professor, Computer Science and Engineering
University of California Santa Cruz
https://www.soe.ucsc.edu/people/mrg

closed 2019-05-23 by mguthaus

Fatsie, 2019-05-23:
I don't get it.
In order to know when the hold timing of the flipflop is violated the STA needs to know when the DOUT of the SRAM becomes 'x' . I don't see how the STA tool can get this data out of the RAM .lib.