VLSIDA/OpenRAM #48: Multi-bank support

robin-tukl, 2019-07-10:
Is it in the current version possible to create a SRAM with more then one bank? I already tried to add "num_banks = 2" to my "myconfig.py" file. But this always results in a layout with only one bank. Also i already saw that the number of banks is not passed to "sram_config". Does that mean that multi bank is not supported yet?

mguthaus, 2019-07-10:
It was supported for a while, but is temporarily removed while we fix some
other issues. I hope to have it added back in the next few weeks or months.
In the interim, you can use two SRAMs and synthesize a mux to select
between the two. It is a little less efficient, but it will work.

Matt

robin-tukl, 2019-07-11:
Thank you for the fast and helpful response.

closed 2019-07-11 by robin-tukl

im-world, 2022-03-26:
I am facing an error while trying to generate a multi-bank memory as well, though it is different from the one described above. The error message I get is
```
Traceback (most recent call last):
  File "/home/openRAM/OpenRAM/compiler/openram.py", line 77, in <module>
    s = sram(sram_config=c,
  File "/home/openRAM/OpenRAM/compiler/sram/sram.py", line 48, in __init__
    self.s.create_netlist()
  File "/home/openRAM/OpenRAM/compiler/sram/sram_base.py", line 197, in create_netlist
    self.add_modules()
  File "/home/openRAM/OpenRAM/compiler/sram/sram_2bank.py", line 56, in add_modules
    self.compute_bus_sizes()
  File "/home/openRAM/OpenRAM/compiler/sram/sram_base.py", line 380, in compute_bus_sizes
    self.num_vertical_line = self.addr_size + self.control_size + log(self.num_banks, 2) + 1
AttributeError: 'sram_2bank' object has no attribute 'control_size'

```

I'm not sure how to resolve this. The environment setup seems to be fine, as I am able to generate single-bank memories. My configuration file is attached below for reference.

```
num_rw_ports    = 1
num_r_ports     = 0
num_w_ports     = 0

word_size       = 32
num_words       = 128
num_banks       = 2
#words_per_row   = 2

tech_name       = "freepdk45"
process_corners = ["TT"]
supply_voltages = [1.1]
temperatures    = [25]

route_supplies  = True
check_lvsdrc    = False

output_path     = "output"
output_name     = "SRAM_32x128_2b_1rw"
instance_name   = "SRAM_32x128_2b_1rw"
```

Thank you :)

mguthaus, 2022-03-26:
We still haven't fixed multiple bank support. We are adjusting the functionality to introduce soft banks which utilize some synthesized Verilog to do the decode and mux logic rather than a hard layout. This should be available soon as we are testing it and taping one out in sky130.

im-world, 2022-03-27:
Thanks for the prompt response! That's great, I'll look forward to it.

AshishDahare, 2022-05-08:
Any update with this? When this feature will be available?

mguthaus, 2022-05-08:
@bugraonal is supposed to merge it in. We are planning a major release shortly.

zaazad, 2023-11-13:
Is this feature currently supported? I attempted to use the default multi-bank (2 bank) example from the repository (scn4m_subm_sram_32kbyte_1rw1r_2x32x4096_8.py) and encountered the following error during make:
 
 ERROR: file simulation.py: line 605: Could not find bl0 net in timing paths.

[openram.modules.pdriver/__init__]: creating pdriver pdriver_7
** Submodules: 66.9 seconds
** Placement: 0.0 seconds

[openram.base.hierarchy_layout/get_bbox]: Size: 7719.400000000001 x 4238.200000000001 with perimeter margin 0
** Routing: 485.6 seconds
** Verification: 0.0 seconds
** SRAM creation: 552.7 seconds
SP: Writing to /tmp/OpenRAM/macros/scn4m_subm_sram_32kbyte_1rw1r_2x32x4096_8/scn4m_subm_sram_32kbyte_1rw1r_2x32x4096_8.sp
[openram.characterizer.functional/__init__]: Random seed for functional simulation: 1699838133832938497
** Spice writing: 0.4 seconds
DELAY: Writing stimulus...
ERROR: file simulation.py: line 605: Could not find bl0 net in timing paths.
Traceback (most recent call last):
  File "/tmp/OpenRAM/sram_compiler.py", line 76, in <module>
    s.save()
  File "/tmp/OpenRAM/compiler/sram.py", line 130, in save
    d.analysis_init(probe_address, probe_data)
  File "/tmp/OpenRAM/compiler/characterizer/delay.py", line 1288, in analysis_init
    self.set_internal_spice_names()
  File "/tmp/OpenRAM/compiler/characterizer/simulation.py", line 520, in set_internal_spice_names
    bl_name_port, br_name_port = self.get_bl_name(self.graph.all_paths, port)
  File "/tmp/OpenRAM/compiler/characterizer/simulation.py", line 624, in get_bl_name
    bl_names.append(self.get_alias_in_path(paths, int_net, cell_mod, exclude_set))
  File "/tmp/OpenRAM/compiler/characterizer/simulation.py", line 605, in get_alias_in_path
    debug.error("Could not find {} net in timing paths.".format(internal_net), 1)
  File "/tmp/OpenRAM/compiler/debug.py", line 48, in error
    assert return_value == 0
AssertionError
make[1]: *** [Makefile:85: scn4m_subm_sram_32kbyte_1rw1r_2x32x4096_8.ok] Error 1
make: *** [Makefile:95: scn4m_subm_sram_32kbyte_1rw1r_2x32x4096_8] Error 2

wjingl, 2025-05-03:
I also encountered this problem and would like to ask you when it was solved