VLSIDA/OpenRAM #49: Could Xyce be used in OpenRAM?

mithro, 2019-07-17:
https://github.com/Xyce/Xyce
> Xyce (zīs, rhymes with "spice") is an open source, SPICE-compatible, high-performance analog circuit simulator, capable of solving extremely large circuit problems by supporting large-scale parallel computing platforms.

Does it make sense to support Xyce as a simulator in OpenRAM?

mguthaus, 2019-07-17:
Absolutely. This has been on my private to-do list for a long time. I've
mostly been lazy due to the dependencies of Xyce. The list is pretty long
and there was no package that I know of...

We currently support ngspice, hspice, xa. ngspice is pretty good for
parallel evaluation of the BSIM models, but it would be interesting to
compare what is best overall.

Matt
---
Matthew Guthaus
Professor
Computer Science & Engineering
University of California Santa Cruz
https://www.soe.ucsc.edu/people/mrg

mguthaus, 2022-07-21:
This was originally added back in 67a67111a6e9274847fe4eb960b1d1fcd5f0a7ab.

closed 2022-07-21 by mguthaus