VLSIDA/OpenRAM #51: Sense Amplifier enable signal

robin-tukl, 2019-09-06:
During my work with the openram layout, I encountered a strange behaviour of the Sense Amplifier enable signal. For me, it looks like the enable signal is only active during the writes (S_en = web and (not csb) and (not clk) ). That means that the output voltage is not stable when you try to read from the SRAM. Is it possible that there is a mistake in the control logic?

mguthaus, 2019-09-06:
Please look at the dev branch as we restructured the control logic to be
more robust to race conditions and to use replica bitline timing that is
part of the main array. This hasn't been pushed to master yet.

In the new branch however, we also only activate on reads with the intent
that you aren't reading while you write. However, we intended to have a
"feed through" write which is not this behavior. I need to look at the
timing and see if this would cause any other issues. It might be just a one
line fix (to use csb instead of web).

Matt

mguthaus, 2019-09-06:
I'm working on this fix, but it may take a while (travelling international
very soon). It turns out our unit test was not checking feed through writes
that were read in the same cycle, so this was not tested in our functional
simulations.

mguthaus, 2019-09-08:
This logic is from an old version, but it was actually correct at the time.
web is active low, so a read would have it be 1 and s_en would be enabled
(1). Please update to get improved timing of control logic -- there were
some other bugs that are fixed in the dev branch.

I was confusing your question with details about writing and reading at the
same time which is another issue and behaves as expected.

Hope that helps.

closed 2019-09-08 by mguthaus