VLSIDA/OpenRAM #52: Lef macro size appears incorrect for freepdk45

tajayi, 2019-09-25:
Hi,
I'm working to generate some rams for use in the OpenROAD project and running into some issues.  Using the provided `compiler/tests/config_freepdk45.py` tests, the generated lef has `SIZE 14.005 BY 38.2` however it has pin port rectangles outside of the macro area. For example
```
         LAYER metal3 ;
         RECT  -10.32 37.24 13.6 37.38 ;
```

Here's an image of what it looks like
[image]

I was able to workaround this by changing the lef to accommodate the full area and then shift the origin
```
  SIZE 24.325 BY 47.7 ;
   ORIGIN 10.32 9.5 ;
```

mguthaus, 2019-09-25:
Which version (commit id) are you using?

mguthaus, 2019-09-25:
Also, what config (just in case)?

tajayi, 2019-09-25:
Sorry using the latest master commit: e86fa8f0b49569adfd17cf63d200e9deb0576f1f
Also using `compiler/tests/config_freepdk45.py` from the source

mguthaus, 2019-09-26:
This fix is in an upcoming release. We had already seen this where it forgot to offset the coordinates. We used the lower left of the bank for relative coordinates and then change it to the lower left of all shapes at the end, but this was commented out for some reason.

closed 2019-09-26 by mguthaus

tajayi, 2019-09-26:
Thanks @mguthaus. Is the fix in any publically available branch that I can try out?

mguthaus, 2019-09-26:
I just pushed this to the public dev branch.