VLSIDA/OpenRAM #54: Add list of papers which use / reference OpenRAM

mithro, 2019-12-03:
It would be good if there was a documentation page which had a list of papers which use OpenRAM or similar.

jeras, 2022-09-17:
This are some of the OpenRam related articles I was reading this week:

Official documentation is a slide presentation, this is probably the most up to date document:
https://docs.google.com/presentation/d/10InGB33N51I6oBHnqpU7_w9DXlx-qe9zdrlco2Yc5co/edit#slide=id.g54b634b348_0_0

Jeffrey T. Butera: OpenRAM: An Open-Source Memory Compiler
https://escholarship.org/uc/item/2vv5q88z

AMC: An Asynchronous Memory Compiler
https://csl.yale.edu/~rajit/ps/amc.pdf
https://github.com/asyncvlsi/AMC

OpenRAM: An Open-Source Memory Compiler
https://escholarship.org/content/qt8x19c778/qt8x19c778_noSplash_b2b3fbbb57f1269f86d0de77865b0691.pdf

mguthaus, 2023-01-13:
This was added to the README. It is updated in 5816c669f3966ba2ec046a87e1ada6a7b7ab6711.