VLSIDA/OpenRAM #55: Figure out why GitHub doesn't auto-detect your license file.

mithro, 2019-12-03:
See https://help.github.com/en/github/creating-cloning-and-archiving-repositories/licensing-a-repository#detecting-a-license and https://licensee.github.io/licensee/

License detection is working for the https://github.com/VLSIDA/puffery repository for example.

mguthaus, 2019-12-03:
I've filed a request about this before and there isn't a good solution. The problem is that there are multiple licenses due to the inclusion of some FreePDK stuff in the tech directory.

mithro, 2019-12-03:
You could move that FreePDK stuff into its own repository?

mguthaus, 2019-12-03:
This is a trade off of working "out of the box" vs a detected licence. I could possibly create a git submodule to get around this though...

mguthaus, 2019-12-03:
Interestingly, it looks like it is detecting it as 3-clause BSD now and I didn't do anything!
[image]

I've still got a request in regarding my testing and code coverage badges though. GitHub is caching them incorrectly even though they are served from my gitlab with a "no cache" header as they instruct us to do.

closed 2019-12-03 by mguthaus

mguthaus, 2019-12-06:
Status badges fixed. These were a permissions problem on my gitlab instance so I just made it public.

mithro, 2019-12-06:
This seems to be actually the problem;
```
tansell@tansell-glaptop:~/github/VLSIDA/OpenRAM$ ~/.gem/ruby/2.5.0/bin/licensee detect LICENSE 
License:        NOASSERTION
Matched files:  LICENSE
LICENSE:
  Content hash:  49d7d639c8d78568b17a301b31162afec7162483
  License:       NOASSERTION
  Closest non-matching licenses:
    BSD-3-Clause-Clear similarity:  92.49%
    BSD-4-Clause similarity:        92.43%
```

It has two very close matches.

mithro, 2019-12-06:
This should be fixed by #56

mithro, 2019-12-07:
Seems to be working now!

[image]
[image]

PR #56 (2019-12-06): Fix LICENSE file.
Match the `BSD 3-Clause "New" or "Revised" License` template that GitHub provides so that license detection actually works correctly.

#### Before;
```
License:        NOASSERTION
Matched files:  LICENSE
LICENSE:
  Content hash:  49d7d639c8d78568b17a301b31162afec7162483
  License:       NOASSERTION
  Closest non-matching licenses:
    BSD-3-Clause-Clear similarity:  92.49%
    BSD-4-Clause similarity:        92.43%
```
#### After;
```
License:        BSD-3-Clause
Matched files:  LICENSE
LICENSE:
  Content hash:  fa22c672927af9c7334874561198799cbf4bdf31
  Attribution:   Copyri [... cut]