VLSIDA/OpenRAM #63: Characterization through SPICE

robin-tukl, 2020-01-15:
Hey everyone,
I have a problem with the specification of power/timing through the spice simulations. Regardless of the size of the generated SRAM, I get the following error.

`
Technology: scn4m_subm
Total size: 32 bits
Word size: 2
Words: 16
Banks: 1
Write size: None
RW ports: 1
R-only ports: 0
W-only ports: 0
Design supply routing skipped. Supplies will have multiple must-connect pins. (route_supplies=True to enable supply routing).
DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
DRC/LVS/PEX is disabled (check_lvsdrc=True to enable).
Performing simulation-based characterization with ngspice
Words per row: 1
Output files are: 
/home/pi/OpenRAM/temp2x16_noana/sram_2_16_scn4m_subm.sp
/home/pi/OpenRAM/temp2x16_noana/sram_2_16_scn4m_subm.v
/home/pi/OpenRAM/temp2x16_noana/sram_2_16_scn4m_subm.lib
/home/pi/OpenRAM/temp2x16_noana/sram_2_16_scn4m_subm.py
/home/pi/OpenRAM/temp2x16_noana/sram_2_16_scn4m_subm.html
/home/pi/OpenRAM/temp2x16_noana/sram_2_16_scn4m_subm.log
/home/pi/OpenRAM/temp2x16_noana/sram_2_16_scn4m_subm.lef
/home/pi/OpenRAM/temp2x16_noana/sram_2_16_scn4m_subm.gds
** Submodules: 1.6 seconds
** Placement: 0.0 seconds
** Routing: 0.1 seconds
** Verification: 0.0 seconds
** SRAM creation: 1.8 seconds
LEF: Writing to /home/pi/OpenRAM/temp2x16_noana/sram_2_16_scn4m_subm.lef
** LEF: 1.0 seconds
GDS: Writing to /home/pi/OpenRAM/temp2x16_noana/sram_2_16_scn4m_subm.gds
** GDS: 0.5 seconds
SP: Writing to /home/pi/OpenRAM/temp2x16_noana/sram_2_16_scn4m_subm.sp
** Spice writing: 0.0 seconds
LIB: Characterizing... 
ERROR: file delay.py: line 1141: Couldn't run a simulation. slew=0.05 load=2.45605
`
Thanks for your help.
Yannick

robin-tukl, 2020-01-15:
It was my mistake. Still had the voltage from free45pdk specified.

closed 2020-01-15 by robin-tukl

abanobemil, 2020-04-21:
Hi, everyone!
I have this error, any idea what could be the problem?
"Traceback (most recent call last):
  File "openram.py", line 77, in <module>
    s.save()
  File "C:\Users\Abano\Desktop\OpenRAM-master\compiler/sram\sram.py", line 108, in save
    lib(out_dir=OPTS.output_path, sram=self.s, sp_file=sp_file)
  File "C:\users\abano\desktop\OpenRAM-master\compiler\characterizer\lib.py", line 34, in __init__
    self.characterize_corners()
  File "C:\users\abano\desktop\OpenRAM-master\compiler\characterizer\lib.py", line 126, in characterize_corners
    self.characterize()
  File "C:\users\abano\desktop\OpenRAM-master\compiler\characterizer\lib.py", line 133, in characterize
    self.compute_delay()
  File "C:\users\abano\desktop\OpenRAM-master\compiler\characterizer\lib.py", line 564, in compute_delay
    char_results = self.d.analyze(probe_address, probe_data, self.slews, self.loads)
  File "C:\users\abano\desktop\OpenRAM-master\compiler\characterizer\delay.py", line 1102, in analyze
    feasible_delays = self.find_feasible_period()
  File "C:\users\abano\desktop\OpenRAM-master\compiler\characterizer\delay.py", line 693, in find_feasible_period
    feasible_delays[self.read_ports[0]] = self.find_feasible_period_one_port(self.read_ports[0])
  File "C:\users\abano\desktop\OpenRAM-master\compiler\characterizer\delay.py", line 650, in find_feasible_period_one_port
    debug.error("Timed out, could not find a feasible period.",2)
  File "C:\users\abano\desktop\OpenRAM-master\compiler\debug.py", line 46, in error
    assert return_value == 0
AssertionError
"
Thanks in advance!

mguthaus, 2020-04-23:
Hi,

Wha is the commit id  of the version you are using?
What memory configuration or test?

This error happens when it can't stimulate to measure delay with a very
long clock period. So, usually something else is wrong.

Matt

abanobemil, 2020-04-23:
Thanks, actually yes it was something else wrong, now it works but when I try to run for a size greater than 16*16 I get the following error:

ERROR: file delay.py: line 824: Sense amp enable timing error. Increase the delay chain through the configuration file.

How should I increase the delay chain and to what value? 

Thanks again.

hznichol, 2020-04-27:
Hello,
You can increase the delay chain by setting this value in the config file
`delay_chain_stages = 9`

Nine stages is the default. You can increase the number of stages to increase the delay.
The delay chain must have an odd number of stages (will error otherwise).
Currently, we don't give a recommended number as it will depend on the configuration.

Hunter

mguthaus, 2020-04-27:
You can also adjust the fanout of the delay chain in the config file:
delay_chain_fanout_per_stage = 4

The default is 4.

Matt