VLSIDA/OpenRAM #67: Error while generating 16 x 16 Array

GyanendraTiwari, 2020-02-16:
I am able to generate 32x32, 256x256, 16x2, etc array successfully but it gives the error while generating 16x16 array.
ERROR: file calibre.py: line 218: control_logic_rw	Geometries: 7526	Checks: 167	Errors: 1
/home/gyanendra/openram/compiler/debug.py(46)error()
assert return_value == 0

Can you please help me in resolving the issue.

Thanks in advance.

mguthaus, 2020-02-16:
Hi,

Are you using the most recent version? Could you try the dev branch?

What is the DRC error? This should be summarized in the temp directory with the extension .DRC.summary and the markers for viewing in .DRC.results.

bkoppelmann, 2020-02-17:
I can reproduce a DRC error for:

```
Technology: freepdk45
Total size: 512 bits
Word size: 8
Words: 64
Banks: 1
Write size: None
RW ports: 1
R-only ports: 0
W-only ports: 0
```

I guess it's the same problem. My control_logic_rw.drc.summary says:
```
RULECHECK Metal1.5 .......... TOTAL Result Count = 1 (1)
```

bkoppelmann, 2020-02-18:
@GyanendraTiwari , does this patch fix your problem?
```
diff --git a/compiler/modules/control_logic.py b/compiler/modules/control_logic.py
index e22d05b3..dfdeb8ca 100644
--- a/compiler/modules/control_logic.py
+++ b/compiler/modules/control_logic.py
@@ -627,7 +627,7 @@ class control_logic(design.design):
         x_offset = self.control_x_offset
 
         x_offset = self.place_util(self.p_en_bar_nand_inst, x_offset, row)        
-        x_offset = self.place_util(self.p_en_bar_driver_inst, x_offset, row)        
+        x_offset = self.place_util(self.p_en_bar_driver_inst, x_offset + 1, row)
 
         self.row_end_inst.append(self.p_en_bar_driver_inst)
 
```

mguthaus, 2022-07-21:
No longer see the error.

closed 2022-07-21 by mguthaus