VLSIDA/OpenRAM #73: Multiple Queries Regarding Simulation of OpenRAM

GyanendraTiwari, 2020-03-05:
Can anyone resolve these three queries?
1. Have you tested the generated memory compiler from OpenRAM in Cadence/Synopsys Tool?
2. If yes are you getting the 10-20 GHz in the said tools as reported by the Openram output file. We have tried compiling the generated schematic in Cadence Virtuoso but we are only able to get frequency up to 400 MHz?
3. In the Config file, we have tried changing the number of Banks but we are getting only one bank.

mguthaus, 2020-03-05:
1. Yes, we have used the lib and lef files with commercial tools.
2. 10-20 GHz is quite high! Where are you seeing this? Is it using simulated output or the analytical models (the default). The analytical models will definitely be inaccurate as they rely on calibration and closed form models.
3. Multiple banks has been temporarily disabled while we add other features. We welcome contributions to add this back.

mguthaus, 2020-03-05:
Please consider joining the slack channel if you have more questions.

closed 2020-03-05 by mguthaus