VLSIDA/OpenRAM #76: Example configurations not working

andresag01, 2020-06-22:
Thanks for a great tool!

I was just giving it a try and I found out that the example configs in the [compiler/example_configs](https://github.com/VLSIDA/OpenRAM/blob/dev/compiler/example_configs/) directory aren't fully working. Specifically, I tried the ones for `scn4m_subm` with `check_lvsdrc = True` (such as [example_config_1rw_1r_scn4m_subm.py](https://github.com/VLSIDA/OpenRAM/blob/dev/compiler/example_configs/example_config_1rw_1r_scn4m_subm.py)) from the current latest `dev` branch (b7c66d7e073566d775ef8b9b902e8931a547e56d) and I get the following errors:

```
ERROR: file magic.py: line 249: sram_1rw_1r_2_16_scn4m_subm     LVS mismatch (results in /home/andresag/Repos/openram_output/snc4m/ports_1rw/analytical/tmp/sram_1rw_1r_2_16_scn4m_subm.lvs.report)
** Verification: 22.5 seconds
** SRAM creation: 36.4 seconds
LEF: Writing to /home/andresag/Repos/openram_test/temp/sram_1rw_1r_2_16_scn4m_subm.lef
** LEF: 0.7 seconds
GDS: Writing to /home/andresag/Repos/openram_test/temp/sram_1rw_1r_2_16_scn4m_subm.gds
** GDS: 0.3 seconds
SP: Writing to /home/andresag/Repos/openram_test/temp/sram_1rw_1r_2_16_scn4m_subm.sp
** Spice writing: 0.1 seconds
LIB: Characterizing... 
ERROR: file magic.py: line 249: sram_1rw_1r_2_16_scn4m_subm     LVS mismatch (results in /home/andresag/Repos/openram_output/snc4m/ports_1rw/analytical/tmp/sram_1rw_1r_2_16_scn4m_subm.lvs.report)
ERROR: file magic.py: line 249: sram_1rw_1r_2_16_scn4m_subm     LVS mismatch (results in /home/andresag/Repos/openram_output/snc4m/ports_1rw/analytical/tmp/sram_1rw_1r_2_16_scn4m_subm.lvs.report)
ERROR: file magic.py: line 249: sram_1rw_1r_2_16_scn4m_subm     LVS mismatch (results in /home/andresag/Repos/openram_output/snc4m/ports_1rw/analytical/tmp/sram_1rw_1r_2_16_scn4m_subm.lvs.report)
```

The generated files show a few errors here and there. For example `setup.tcl` and `sram_1rw_1r_2_16_scn4m_subm.lvs.err` contain:

```
# Setup file for netgen
ignore class c
equate class {-circuit1 nfet} {-circuit2 n}
equate class {-circuit1 pfet} {-circuit2 p}
# We must flatten these because the ports are disconnected
flatten class {-circuit1 dummy_cell_6t}
flatten class {-circuit1 dummy_cell_1rw_1r}
flatten class {-circuit1 dummy_cell_1w_1r}
flatten class {-circuit1 bitcell_array_0}
flatten class {-circuit1 pbitcell_0}
flatten class {-circuit1 pbitcell_1}
property {-circuit1 nfet} remove as ad ps pd
property {-circuit1 pfet} remove as ad ps pd
property {-circuit2 n} remove as ad ps pd
property {-circuit2 p} remove as ad ps pd
permute transistors
```

```
Error setup.tcl:2 (ignoring), No such cell!
Error setup.tcl:6 (ignoring), No such cell or bad file number!
Error setup.tcl:8 (ignoring), No such cell or bad file number!
Error setup.tcl:10 (ignoring), No such cell or bad file number!
Error setup.tcl:11 (ignoring), No such cell or bad file number!
```

The file `sram_1rw_1r_2_16_scn4m_subm.lvs.report` also contains a lot of mismatch errors such as this:

```
Cell write_driver disconnected node: comment_0_0#

Subcircuit summary:
Circuit 1: write_driver                    |Circuit 2: write_driver
-------------------------------------------|-------------------------------------------
nfet (9)                                   |n (9)
pfet (7)                                   |p (7)
Number of devices: 16                      |Number of devices: 16
Number of nets: 16 **Mismatch**            |Number of nets: 13 **Mismatch**
---------------------------------------------------------------------------------------
NET mismatches: Class fragments follow (with fanout counts):
Circuit 1: write_driver                    |Circuit 2: write_driver

---------------------------------------------------------------------------------------
```

I am using the Ubuntu docker hub image linked from the README.

Do you know what causes these issues?

Thanks for the help!

mguthaus, 2020-06-22:
Hi Andres,

I just ran this myself and it passed. I am no longer using that Docker and have upgraded to Magic 8.3, so that might be an issue. The error looks like there is a short circuit in the layout.  

The "errors" during LVS are expected as we hard code some flatten statements to get rid of LVS symmetry problems. There should be a more elegant way to do this, but we just haven't yet.

andresag01, 2020-06-23:
Hi @mguthaus, you are right! Thanks for the help. I updated the Magic version to 8.3.27 and it works without problems. Perhaps the issue is that the documentation at README.md is out-of-date? I create a PR to address it: https://github.com/VLSIDA/OpenRAM/pull/77

andresag01, 2020-06-23:
@mguthaus: Just one follow up question. I am getting the following error when compiling an SRAM with the example config [example_config_1rw_1r_scn4m_subm.py](https://github.com/VLSIDA/OpenRAM/blob/dev/compiler/example_configs/example_config_1rw_1r_scn4m_subm.py) with `analytical_delay = False` (instead of the default `True`):

```
LIB: Characterizing... 
[characterizer.lib/prepare_tables]: Loads: [ 2.45605  9.8242  39.2968 ]
[characterizer.lib/prepare_tables]: Slews: [0.0125 0.05   0.4   ]
[characterizer.lib/characterize_corners]: Characterizing corners: [('TT', 5.0, 25), ('FF', 5.0, 25), ('SS', 5.0, 25)]
[characterizer.lib/characterize_corners]: Corner: ('TT', 5.0, 25)
[characterizer.lib/characterize_corners]: Writing to /home/andresag/Repos/openram_experiments/tests/sram_rw1_r1_w0/proper/temp/sram_2_16_scn4m_subm_TT_5p0V_25C.lib
[characterizer.delay/find_feasible_period_one_port]: Trying feasible period: 10.0ns on Port 0
[characterizer.delay/find_feasible_period_one_port]: Trying feasible period: 10.0ns on Port 1
[characterizer.delay/find_feasible_period_one_port]: Trying feasible period: 20.0ns on Port 1
[characterizer.delay/find_feasible_period_one_port]: Trying feasible period: 40.0ns on Port 1
[characterizer.delay/find_feasible_period_one_port]: Trying feasible period: 80.0ns on Port 1
[characterizer.delay/find_feasible_period_one_port]: Trying feasible period: 160.0ns on Port 1
[characterizer.delay/find_feasible_period_one_port]: Trying feasible period: 320.0ns on Port 1
[characterizer.delay/find_feasible_period_one_port]: Trying feasible period: 640.0ns on Port 1
[characterizer.delay/find_feasible_period_one_port]: Trying feasible period: 1280.0ns on Port 1
ERROR: file delay.py: line 661: Timed out, could not find a feasible period.
> /home/andresag/Programs/OpenRAM/compiler/debug.py(46)error()
-> assert return_value == 0
(Pdb) 
```

I was initially wondering why openram was taking so long to terminate, so I ran it with `debug_level = 1` to get the output above. As you can see, it appears that it failed to characterize and then it ended up on a prompt. Does this work for you? How can I get around the problem?

For reference, the config file is:

```Python
word_size = 2                                                                       
num_words = 16                                                                      
                                                                                    
tech_name = "scn4m_subm"                                                            
nominal_corners_only = False                                                        
process_corners = ["TT"]                                                            
supply_voltages = [5.0]                                                             
temperatures = [25]                                                                 
                                                                                    
num_rw_ports = 1                                                                    
num_r_ports = 1                                                                     
num_w_ports = 0                                                                     
                                                                                    
route_supplies = True                                                               
check_lvsdrc = True                                                                 
                                                                                    
analytical_delay = False                                                            
                                                                                    
output_path = "temp"                                                                
output_name = "sram_{0}_{1}_{2}".format(word_size,                                  
                                        num_words,                                  
                                        tech_name)                                  
debug_level = 1                                                                     
                                                                                    
purge_temp = False
```

The problem also happens when:

```
num_rw_ports = 0                                                                    
num_r_ports = 1                                                                     
num_w_ports = 1
```

andresag01, 2020-07-03:
The problem was fix with the latest commits in `dev`. So i will close this issue.

closed 2020-07-03 by andresag01