VLSIDA/OpenRAM #78: Availability of an OpenROM

PPlinux, 2020-06-26:
Dear,
I'm looking for a open source small OpenRom e.g. 256Words 16bit for a custom design.
Has OpenRam team worked on ROM or did the OpenRam team know the availability 
I touth Aliance had done some work but I believe its outdated and not loger supported.
Hoping to receive a positive reaction, I meanwhile remain,
Patrick Pelgrims

mguthaus, 2020-06-26:
Hi Patrick,

I think you asked this on the mail list too, but you weren't subscribed to receive a reply. 

We do not support ROMs, but this would be a great project using the OpenRAM infrastructure.

Matt

closed 2020-06-26 by mguthaus

PPlinux, 2020-06-26:
Dear,

Can we adapt OpenRam to function as ROM ?

Greetings,
Patrick

mguthaus, 2020-06-26:
Yes, that would be a great project.