VLSIDA/OpenRAM #79: Failed to analyze multiport memory with RW:1 R:1

andresag01, 2020-06-26:
I am having problems compiling a multiport SRAM with 1 read-write port and 1 read-only port. I am using the following config:

```
word_size = 2                                                                       
num_words = 16                                                                      
                                                                                    
num_rw_ports = 1                                                                    
num_r_ports = 1                                                                     
num_w_ports = 0                                                                     
                                                                                    
tech_name = "freepdk45"                                                             
nominal_corners_only = False                                                        
process_corners = ["TT"]                                                            
supply_voltages = [1.0]                                                             
temperatures = [25]                                                                 
                                                                                    
route_supplies = False                                                              
check_lvsdrc = False                                                                
                                                                                    
analytical_delay = False                                                            
                                                                                    
purge_temp = False                                                                  
                                                                                    
output_path = "."                                                                   
output_name = "sram_{0}_{1}_{2}".format(word_size, num_words, tech_name)                                                                                            
```

The problem is that the function [parse_spice_list()](https://github.com/VLSIDA/OpenRAM/blob/master/compiler/characterizer/charutils.py#L18) is failing to parse the `delay_sen1` measurement from the `tmp/timing.lis file`, i.e the output file from ngspice, when analysing the delays for read port 1. This is the error message in `timing.lis`:

```
Error: measure  delay_sen1  (TARG) : out of interval
.meas tran delay_sen1 trig v(clk1) val=0.5 fall=1 td=27.5n targ v(xsram_2_16_freepdk45.s_en0) val=0.5 rise=1 td=27.5n failed!
```

Unfortunately, the error was not caught earlier and [parse_spice_list()](https://github.com/VLSIDA/OpenRAM/blob/master/compiler/characterizer/charutils.py#L18) continuously returns the string `"Failed"` rather than a floating-point number. This then causes the [try_period()](https://github.com/VLSIDA/OpenRAM/blob/master/compiler/characterizer/delay.py#L998) function to repeatedly fail, which in turn, forces [find_min_period_one_port()](https://github.com/VLSIDA/OpenRAM/blob/master/compiler/characterizer/delay.py#L973) into an infinite loop until a timeout occurs [here](https://github.com/VLSIDA/OpenRAM/blob/master/compiler/characterizer/delay.py#L975).

Do you know how I can get around the problem with `ngspice`? Thanks!

mguthaus, 2020-06-26:
Hi Andres,

That error means that the delay_sen1 signal is not transitioning properly during the simulation time. 

Depending on where you are in characterization, this is called twice. 

 In our characterizer, it will double the period until it finds a feasible time or it will time-out after so many times (9). This finds the "feasible" period. 

After this, it tries to find the min period with a bisection search using the above feasible period and another infeasible period. This should converge naturally, but it has a limit of 25 times "just in case". In practice, that shouldn't happen.

Does your unit test 21_ngspice_delay_test.py pass? This is essentially the same as that config but it has word_size=1 and a single port.

Matt

andresag01, 2020-06-26:
Hi Matt,

Thanks for the quick reply. I ran the unit test you suggested using `python3 21_ngspice_delay_test.py -n -t freepdk45 -v` and got a pass. However, this tests memories with one port and I can generate those without any problems. I can also generate memories with 1 read-only and 1 write-only port, but I get the error I described above when generating memories with 1 read-write port and 1 read-only port.

>After this, it tries to find the min period with a bisection search using the above feasible period and another infeasible period. This should converge naturally, but it has a limit of 25 times "just in case". In practice, that shouldn't happen.

I think this is the bit that is failing for me. I do not know how it works in detail, but I think the read ports are characterized one-by-one. It seems that the tool can characterize port 0 without problems, but ngspice fails when trying to run for port 1.

I noticed that ngspice does not return a failure code when run in batch mode despite finding errors. Therefore, OpenRAM thinks it worked and tries to use some regular expressions to parse the output. This parsing always fails and the convergence never happens, so OpenRAM times out after 25 times.

Please let me know what else I can do to help. I tried looking more closely at the code, but I am no expert...

Thanks,
Andres

mguthaus, 2020-06-26:
@hznichol can you take a look at this?

mguthaus, 2020-07-03:
@hznichol fixed this in 119bd946

closed 2020-07-03 by mguthaus

andresag01, 2020-07-03:
Thanks for the help! I just ran the test and it appears to work!