VLSIDA/OpenRAM #83: calibre pex: faulty timing

bvhoof, 2020-07-30:
In calibre pex mode, the timing cannot be executed correctly:

the measurement statements in the stim.sp file, ran on the pexed file do not work:
for instance: tran v_bl_READ_ZERO0 FIND v(xsram_2_16_freepdk45.xbank0.bl_1) AT=4000.0n 

The target is a a node in a subcircuit. However, after PEX, the netlist is flattened and this node is not available as written in the stimulus file.
Two options are possible: 
1) The node is available as 'flattened node', so the stimulus writer file needs to be pex-aware and change the net name
2) The calibre pex needs to be ran in 'hierarchical mode', so that the hierarchy is preserved (no idea if this is possible in calibre, I know from experience that this is possible in quantus QRC)

jcirimel, 2020-07-30:
Hi bvhoof,

Thanks for letting us know about this; we have only supported pex mode using Magic for extraction and with the scn4m_subm technology so far. Of the two options you have mentioned, we chose option 1. We label the nodes that we wish to access post extraction on the physical layout so that the names are preserved in the extracted spice. If you need a quick fix this approach might also work with Calibre, otherwise thanks for putting this on our radar.

bvhoof, 2020-07-30:
Hi Jesse,

Ok, I will try to hotfix this myself and then create a pull request. Can you point me in the right direction? Where do you generate the stim.sp file?

FYI, to re-create this, you will probably need to pull my other hotfix, so that calibre actually runs.

Thanks in advance,
Bob.

jcirimel, 2020-07-30:
Hi Bob,

The stimulus writing is done with compiler/characterizer/stimuli.py and compiler/characterizer/delay.py. If you add nets to the stimulus sram file circuit definition, remember you must ensure they match in the top level sram spice circuit definition and are ordered the same. I realize now that the changes which added pex support were ever adding to the public OpenRAM repo, including the code to add the labels to preserve names and alter the stimulus files. I will try to get this merged in tomorrow.

Thanks,
Jesse

bvhoof, 2020-07-30:
Hi Jesse,

Thanks for the work. I will look at it after the merge. I suppose it will be on the dev branch?

Best,
Bob.

mguthaus, 2020-07-30:
Yes, it will be in the dev branch.

bvhoof, 2020-08-13:
Hi Jesse,

I still need a little more hints. I expected to find some of your code, based on the magic pex script, which would replace the names.

I found the structure of the calls: 
The analysis_init calls
- set_internal_spice_names, which calls:
-- get_bl_name in the delay.py (and get_sen_name) to generate the bl names and sen name
- create_measurement_objects, which calls:
-- create_read_port_measurement_objects, which calls:
--- create_bitline_measurement_objects, to create measurement statements, based on the previously generated bl names
--- create_debug_measurement_objects, to create measurement statements, based on the prevously generated sen name
--- create_read_bit_measures, which calls
---- get_bit_measures, to create measurement statements, based on the call:
----- self.sram.get_cell_name, which outputs cell_name and cell_inst, which get modified using:
----- cell_inst.mod.get_storage_net_names

However, none of these low-level functions (get_bl_name, get_sen_name, get_cell_name and get_storage_net_names) seem to be pex-aware. Is there an example file that is made pex-aware?

Best regards,
Bob

jcirimel, 2020-08-18:
Hi Bob, 
Sorry for the delay, the pex aware code is now in the dev branch. Each bank now has a function "reverse_transformation_bitcell" which gets the coordinates of bitcell Q and Q_bar nodes as well as bitlines. This means the bitcell layout needs to be labeled appropriately. This information is used in sram_base.py to add top level labels on necessary nodes for preservation during pex. In the extracted spice file and in the stimulus file, we edit the top level sram definition to include all pins needed for simulation. This is done in magic.py, so you could add something similar to the calibre extraction script. OpenRAM can be run with pex mode with the -b -c flags. Here is an example of pex sram and stimulus spice file pairs generated from example_config_scn4m_subm.py
[example_scmos.zip](https://github.com/VLSIDA/OpenRAM/files/5093250/example_scmos.zip)

closed 2020-11-12 by mguthaus