VLSIDA/OpenRAM #89: Not able to use multiple banks

jkhan810, 2020-09-09:
I get the attribute error and the script stops there. If I use one bank, it runs smoothly, but I want to create larger memory.  Do I have to set the attribute for object manually? 

Technology: FreePDK45
Total size: 512 bits
Word size: 8
Words: 32
Banks: 2
Write size: None
RW ports: 1
R-only ports: 0
W-only ports: 0
Design supply routing skipped. Supplies will have multiple must-connect pins. (route_supplies=True to enable supply routing).
DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
DRC/LVS/PEX is disabled (check_lvsdrc=True to enable).
Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
Only characterizing nominal corner.
Words per row: 1
Output files are: 
/calypto/technology_group/jkhan/openram/run2/sram_8_32_FreePDK45.lvs
/calypto/technology_group/jkhan/openram/run2/sram_8_32_FreePDK45.sp
/calypto/technology_group/jkhan/openram/run2/sram_8_32_FreePDK45.v
/calypto/technology_group/jkhan/openram/run2/sram_8_32_FreePDK45.lib
/calypto/technology_group/jkhan/openram/run2/sram_8_32_FreePDK45.py
/calypto/technology_group/jkhan/openram/run2/sram_8_32_FreePDK45.html
/calypto/technology_group/jkhan/openram/run2/sram_8_32_FreePDK45.log
/calypto/technology_group/jkhan/openram/run2/sram_8_32_FreePDK45.lef
/calypto/technology_group/jkhan/openram/run2/sram_8_32_FreePDK45.gds
**Traceback (most recent call last):
  File "/home/jkhan/openram/compiler/openram.py", line 77, in <module>
    s = sram(sram_config=c,
  File "/home/jkhan/openram/compiler/**sram/sram.py", line 44, in __init**__
    self.s.create_netlist()
  File "/home/jkhan/openram/compiler/sram/sram_base.py", line 96, in create_netlist
    self.add_modules()
  File "/home/jkhan/openram/compiler/sram/sram_2bank.py", line 61, in add_modules
    self.compute_bus_sizes()
  File "/home/jkhan/openram/compiler/sram/sram_base.py", line 173, in compute_bus_sizes
    self.num_vertical_line = self.addr_size + self.control_size + log(self.num_banks, 2) + 1
AttributeError: 'sram_2bank' object has no attribute 'control_size'**

mguthaus, 2020-09-09:
The multibank options has been temporarily disabled. Until then, your best option is to create a Verilog wrapper to utilize multiple SRAMs.

We will get it added back hopefully this fall.

closed 2020-09-09 by mguthaus

jkhan810, 2020-09-09:
@mguthaus Thanks a lot for response!

jkhan810, 2020-09-10:
Hello @mguthaus , I have a quick question. Do we have an option for step size in Openram? Like, If I want to go from min bit to max bit, can I go in steps?

mguthaus, 2020-09-10:
I don't think the issues is a good place for questions. You can get help either on the mail list or the Slack channel. See the README for information on these.

jkhan810, 2020-09-10:
Okay , Thanks!