VLSIDA/OpenRAM #95: If tool isn't found, openram.py doesn't terminate, it just gets stuck.

mithro, 2020-11-03:
```
python3 /home/tim/github/VLSIDA/OpenRAM/compiler/openram.py -v -p riscv-sky130-1kbyte -o riscv-sky130-1kbyte configs/riscv-sky130-1kbyte.py
ERROR: file globals.py: line 262: calibre not found. Cannot find DRC tool.
<nothing happens here>
```

I would expect it to exit with a non-zero return code.

mithro, 2020-11-03:
I think this is happening because your Makefile is doing;
```
python3 $(OPENRAM_HOME)/openram.py $(OPTS) -p $(bname) -o $(bname) $< 2>&1 > /dev/null
```
?

mguthaus, 2020-11-03:
Can you run it in verbose mode "-v"? I'm guessing that it is actually routing, not stuck.

mithro, 2020-11-03:
So that isn't an error, it is a warning?

mguthaus, 2020-11-03:
I would classify it as an error. If it cannot check the design, it is broken.

mithro, 2020-11-03:
I would normally say that a tool should stop with a non-zero exit code after a fatal error (generating incorrect output seems like a fatal error to me :-).

mguthaus, 2020-11-03:
Fixed in 6335bc37849ea906e5b2d37cc205535b2fe88b22

closed 2020-11-03 by mguthaus