VLSIDA/OpenRAM #96: metal layer names in .lef

sheriefreda, 2020-11-09:
When I use freepdk45 as my technology, the .lef for the memory module has the layers defined as m1, m2, ...etc, whereas the the freepdk45 defines the layers as metal1, metal2, ...etc, as a result I get errors when I try to do place and route with Innovus. Here is an example

**ERROR: (IMPLF-53):	The layer 'm2' referenced in pin 'din0[0]' in macro 'sram_8_16_freepdk45' is not found in the database. A layer must be defined in the LEF technology LAYER section before it can be referenced from a macro. Review the LEF files specified in the init_lef_file variable to see if the layer does not exist or is specified after the one that defines the macro.

mguthaus, 2020-11-09:
Thanks, we will fix this shortly. We shortened the names internally but didn't test the LEF files since then.

mguthaus, 2020-11-09:
Fixed in 66633a843bb1e8d113339be036e5b5fab0e07866

closed 2020-11-09 by mguthaus

sheriefreda, 2020-11-10:
thanks a lot. working now.