VLSIDA/OpenRAM #107: Operating conditions in corners are wrong

mguthaus, 2021-01-04:
Regarding liberty opcond, the problem is that some models use 1.9 in operating condition, but that is different to the voltage map, in order to support multi-voltage implementation, some tools rely on pg-pin voltage for linking, while others non MV-aware just assume single voltage by using operating condition alone. To avoid linking inconsistency, these should match in libraries modeling pg-pins.
sram_1rw1r_32_256_8_sky130_TT_1p9V_25C.lib

operating_conditions(OC){
    process : 1.0 ;
    voltage : 1.9 ;
    temperature : 25;
    }
...
    nom_voltage : 1.8;
    nom_temperature : 25;
    nom_process : 1.0;
...
    voltage_map ( VDD, 1.8 );
    voltage_map ( GND, 0 );

mguthaus, 2021-01-19:
Any update on this?

hznichol, 2021-01-20:
I removed the nominal voltage/temp/process and made the voltage map the same as the operating conditions.
Was going to test at least the syntax before pushing the change.

mguthaus, 2021-01-20:
Go ahead and push. You can assume the previous syntax was correct and just the values were wrong.

mguthaus, 2021-02-10:
What is the status on this?

hznichol, 2021-02-26:
Change has been added to which now has "operating_conditions" and "nom_voltage, nom_temperature, nom_process" as the same values in the lib file.

closed 2021-02-26 by hznichol