VLSIDA/OpenRAM #108: Error occurred while determining bitline names. Can cause faults in simulation.

mguthaus, 2021-01-19:
The code gives the following warning:
Error occurred while determining bitline names. Can cause faults in simulation.

Is this a problem? Should it be fixed?

commit: 

608e4b81f1763727e7efe087d591c76956869fe6

config:
`#Data word size
word_size = 32
# Number of words in the memory
num_words = 16

# Technology to use in $OPENRAM_TECH
tech_name = "freepdk45"

nominal_corner_only = False
# Process corners to characterize
process_corners = ["TT"]
# Voltage corners to characterize
supply_voltages = [ 1.2 ]
# Temperature corners to characterize
temperatures = [ 25 ]

route_supplies = False
check_lvsdrc = False

load_scales = [0.5, 1, 4]
slew_scales = [0.5, 1]

# Port options
num_rw_ports=1
num_r_ports=0
num_w_ports=0
# Output directory for the results
output_path = "Icache_results"
# Output file base name
output_name = "ICache_{0}_{1}_{2}".format(word_size,num_words,tech_name)

# Disable analytical models for full characterization (WARNING: slow!)
analytical_delay = False

# To force this to use magic and netgen for DRC/LVS/PEX
# Could be calibre for FreePDK45
drc_name = "magic"
lvs_name = "netgen"`

mguthaus, 2021-02-10:
What is the status on this?

hznichol, 2021-07-29:
Fix will be applied in pending commit f25dcf1b6372670f88c7129a564b3959c31a7e4f

closed 2021-07-29 by mguthaus