VLSIDA/OpenRAM #109: 30_openram_back_end_test using calibre and freepdk45

bvhoof, 2021-02-03:
commit: b3e249c722d33

although overriding the drc/lvs/pex tools to calibre, it still errors on: ERROR: file globals.py: line 253: magic not found
If I insert print statements before, it prints 'calibre' and the correct path.

Any ideas?

Best regards,
Bob.

mguthaus, 2021-02-03:
I'm not able to replicate this -- it works fine on my computer. How are you running the test? 
```
$ ./30_openram_back_end_test.py -t freepdk45
 ______________________________________________________________________________
|==============================================================================|
|=========                     Running Test for:                      =========|
|=========                         freepdk45                          =========|
|=========               ./30_openram_back_end_test.py                =========|
|=========                       /home/mrg/temp                       =========|
|==============================================================================|
WARNING: file 30_openram_back_end_test.py: line 51: Failed to find coverage installation. This can be installed with pip3 install coverage
.
----------------------------------------------------------------------
Ran 1 test in 14.234s

OK
```

closed 2021-02-03 by mguthaus

bvhoof, 2021-02-04:
I ran this exactly the same as you. I run in an anaconda environment with all correct stuff installed. I installed magic and netgen (which I don't really need) Now, it goes a bit further into the code, but it still crashes on the same issue. 

For some reason, it thinks that magic is the DRC tool: 
If I print OPTS.drc_exe in the verify/__init__.py, it prints the correct tool: calibre
If i print the inspection in the base/hierarchy_design.py right under the import verify statement of the DRC_LVS() definition:
```
import verify
import inspect
print(inspect.getmodule(verify.run_drc))
```
It does not print that, it crashes beforehand, altough it says in the stack trace that it got further than that statement.
```
./30_openram_back_end_test.py -t freepdk45
 ______________________________________________________________________________ 
|==============================================================================|
|=========                     Running Test for:                      =========|
|=========                         freepdk45                          =========|
|=========               ./30_openram_back_end_test.py                =========|
|=========                     /tmp/openram_debug                     =========|
|==============================================================================|
__init__.py drc_exe:calibre
__init__.py lvs_exe:calibre
__init__.py pex_exe:calibre
ERROR: file magic.py: line 220: Unable to find the total error line in Magic output.
Traceback (most recent call last):
  File "/users/micas/bvanhoof/Documents/projects/OpenRAM/compiler/openram.py", line 77, in <module>
    s = sram(sram_config=c,
  File "/users/micas/bvanhoof/Documents/projects/OpenRAM/compiler/sram/sram.py", line 55, in __init__
    self.s.create_layout()
  File "/users/micas/bvanhoof/Documents/projects/OpenRAM/compiler/sram/sram_base.py", line 213, in create_layout
    self.DRC_LVS(final_verification=OPTS.route_supplies, force_check=OPTS.check_lvsdrc)
  File "/users/micas/bvanhoof/Documents/projects/OpenRAM/compiler/base/hierarchy_design.py", line 64, in DRC_LVS
    self.drc_errors = verify.run_drc(self.cell_name, tempgds, tempspice, extract=True, final_verification=final_verification)
  File "/users/micas/bvanhoof/Documents/projects/OpenRAM/compiler/verify/magic.py", line 220, in run_drc
    debug.error("Unable to find the total error line in Magic output.", 1)
  File "/users/micas/bvanhoof/Documents/projects/OpenRAM/compiler/debug.py", line 47, in error
    assert return_value == 0
AssertionError
Traceback (most recent call last):
  File "/users/micas/bvanhoof/anaconda3/envs/openram/lib/python3.8/unittest/case.py", line 60, in testPartExecutor
    yield
  File "/users/micas/bvanhoof/anaconda3/envs/openram/lib/python3.8/unittest/case.py", line 676, in run
    self._callTestMethod(testMethod)
  File "/users/micas/bvanhoof/anaconda3/envs/openram/lib/python3.8/unittest/case.py", line 633, in _callTestMethod
    method()
  File "./30_openram_back_end_test.py", line 69, in runTest
    self.assertEqual(os.path.exists(filename), True)
  File "/users/micas/bvanhoof/anaconda3/envs/openram/lib/python3.8/unittest/case.py", line 912, in assertEqual
    assertion_func(first, second, msg=msg)
  File "/users/micas/bvanhoof/anaconda3/envs/openram/lib/python3.8/unittest/case.py", line 905, in _baseAssertEqual
    raise self.failureException(msg)
AssertionError: False != True
```

Can you point me in a direction how this might happen?

Thanks,
Bob