VLSIDA/OpenRAM #111: freepdk using calibre lvs: top level ports issue

bvhoof, 2021-02-09:
Hi,

Recently, you added labels on the top level (for pex and timing).
However, for calibre LVS, there is a spice netlist without propagated ports and the gds (and extracted spice) do have all these propagated ports.

Is there some kind of switch to propagate the ports in spice, similar to the layout, so that LVS is clean?

Best,
Bob

some code below:
run script:
 `python3 $OPENRAM_HOME/openram.py myconfig --debug`
config file:
 ```
# Data word size
word_size = 2
# Number of words in the memory
num_words = 16
# temporary fix:
#words_per_row = 1

# Technology to use in $OPENRAM_TECH
tech_name = "freepdk45"
# Process corners to characterize
process_corners = ["TT"]
# Voltage corners to characterize
supply_voltages = [ 0.6 ]
# Temperature corners to characterize
temperatures = [ 25 ]

# Output directory for the results
output_path = "/tmp/openram"
# Output file base name
output_name = "sram_{0}_{1}_{2}".format(word_size,num_words,tech_name)

# Disable analytical models for full characterization (WARNING: slow!)
# analytical_delay = False

# To force this to use magic and netgen for DRC/LVS/PEX
# Could be calibre for FreePDK45
#drc_name = "calibre"
#lvs_name = "calibre"
#pex_name = "calibre"
threads = 8

use_pex = True
check_lvsdrc = True
analytical_delay = False

debug_level = 10
```
generated spice netlist sram subckt definition:
```
.SUBCKT sram_2_16_freepdk45 din0[0] din0[1] addr0[0] addr0[1] addr0[2] addr0[3] csb0 web0 clk0 dout0[0] dout0[1] vdd gnd
*.PININFO din0[0]:I din0[1]:I addr0[0]:I addr0[1]:I addr0[2]:I addr0[3]:I csb0:I web0:I clk0:I dout0[0]:O dout0[1]:O vdd:B gnd:B
* INPUT : din0[0] 
* INPUT : din0[1] 
* INPUT : addr0[0] 
* INPUT : addr0[1] 
* INPUT : addr0[2] 
* INPUT : addr0[3] 
* INPUT : csb0 
* INPUT : web0 
* INPUT : clk0 
* OUTPUT: dout0[0] 
* OUTPUT: dout0[1] 
* POWER : vdd 
* GROUND: gnd 
....
.ENDS sram_2_16_freepdk45
```
spice netlist from calibre:
```
.SUBCKT sram_2_16_freepdk45 s_en0 w_en0 p_en_bar0 wl_en0 clk_buf0 bitcell_Q_b0_r0_c0 bitcell_Q_b0_r1_c0 bitcell_Q_b0_r2_c0 bitcell_Q_b0_r3_c0 bitcell_Q_b0_r4_c0 bitcell_Q_b0_r5_c0 bitcell_Q_b0_r6_c0 bitcell_Q_b0_r7_c0 bitcell_Q_b0_r8_c0 bitcell_Q_b0_r9_c0 bitcell_Q_b0_r10_c0 bitcell_Q_b0_r11_c0 bitcell_Q_b0_r12_c0 bitcell_Q_b0_r13_c0 bitcell_Q_b0_r14_c0
+ bitcell_Q_b0_r15_c0 bitcell_Q_bar_b0_r0_c0 bitcell_Q_bar_b0_r1_c0 bitcell_Q_bar_b0_r2_c0 bitcell_Q_bar_b0_r3_c0 bitcell_Q_bar_b0_r4_c0 bitcell_Q_bar_b0_r5_c0 bitcell_Q_bar_b0_r6_c0 bitcell_Q_bar_b0_r7_c0 bitcell_Q_bar_b0_r8_c0 bitcell_Q_bar_b0_r9_c0 bitcell_Q_bar_b0_r10_c0 bitcell_Q_bar_b0_r11_c0 bitcell_Q_bar_b0_r12_c0 bitcell_Q_bar_b0_r13_c0 bitcell_Q_bar_b0_r14_c0 bitcell_Q_bar_b0_r15_c0 bitcell_Q_b0_r0_c1 bitcell_Q_b0_r1_c1 bitcell_Q_b0_r2_c1
+ bitcell_Q_b0_r3_c1 bitcell_Q_b0_r4_c1 bitcell_Q_b0_r5_c1 bitcell_Q_b0_r6_c1 bitcell_Q_b0_r7_c1 bitcell_Q_b0_r8_c1 bitcell_Q_b0_r9_c1 bitcell_Q_b0_r10_c1 bitcell_Q_b0_r11_c1 bitcell_Q_b0_r12_c1 bitcell_Q_b0_r13_c1 bitcell_Q_b0_r14_c1 bitcell_Q_b0_r15_c1 bitcell_Q_bar_b0_r0_c1 bitcell_Q_bar_b0_r1_c1 bitcell_Q_bar_b0_r2_c1 bitcell_Q_bar_b0_r3_c1 bitcell_Q_bar_b0_r4_c1 bitcell_Q_bar_b0_r5_c1 bitcell_Q_bar_b0_r6_c1
+ bitcell_Q_bar_b0_r7_c1 bitcell_Q_bar_b0_r8_c1 bitcell_Q_bar_b0_r9_c1 bitcell_Q_bar_b0_r10_c1 bitcell_Q_bar_b0_r11_c1 bitcell_Q_bar_b0_r12_c1 bitcell_Q_bar_b0_r13_c1 bitcell_Q_bar_b0_r14_c1 bitcell_Q_bar_b0_r15_c1 vdd gnd addr0[0] addr0[1] addr0[2] addr0[3] din0[0] din0[1] bl0_0 br0_0 bl0_1
+ br0_1 csb0 web0 clk0 dout0[0] dout0[1]
...
.ENDS
```

mguthaus, 2021-02-10:
Hi, we will take a look at this. It should not be using that netlist for LVS but just for simulation.

bvhoof, 2021-03-01:
Hi,

I solved it myself by creating a new hierarchical pex extractor script in calibre.
Now, it does a hierarchical pex extract. This way, the module names stay the same as in the generated spice netlist. Additionally, simulation runs faster.
Then, I modified all the code so that if calibre pex is used, it runs as if it was a hierarchical non-pexed design.

See pull request 110 for all calibre related fixes.
I hope you can merge this.

Regression test works, although I had to re-write test 30 to pass trough the technology flag correctly.

Best regards,
Bob.

mguthaus, 2021-03-02:
Merged in your changes.

closed 2021-03-02 by mguthaus