VLSIDA/OpenRAM #112: Generating memory with 1R 1W port throws an error on Master branch

aparna1996, 2021-02-27:
When I run with 1RW port, the run completes successfully. But when I run with 1R and 1W port I get the below error. (This is with freepdk45 technology)

`  File "/misc/scratch/akumari/thesis/OpenRAM/compiler/characterizer/delay.py", line 1363, in analytical_delay
    debug.error("Measurement name not recognized: {}".format(mname), 1)`

The above error is when I run with analytical_delay = True

When I instead run with analytical_delay = False, I get the below error
` File "/misc/scratch/akumari/thesis/OpenRAM/compiler/characterizer/delay.py", line 325, in <listcomp>
  mod_type_bools = [mod.is_non_inverting() for mod in edge_mods]
AttributeError: 'replica_bitcell_2port' object has no attribute 'is_non_inverting'`

My inputs are: 
**Config file:** 
[OpenRAM_1r1w_py.txt](https://github.com/VLSIDA/OpenRAM/files/6053432/OpenRAM_1r1w_py.txt)

**Error Log with analytical_delay = True**
[OpenRAM_1r1w_modelon.log](https://github.com/VLSIDA/OpenRAM/files/6053429/OpenRAM_1r1w_modelon.log)
**Error Log with analytical_delay = False**
[OpenRAM_1r1w_modeloff.log](https://github.com/VLSIDA/OpenRAM/files/6053430/OpenRAM_1r1w_modeloff.log)

mguthaus, 2021-07-29:
@hznichol Can you fix this ASAP?

hznichol, 2021-07-29:
Fix will be applied in pending commit b408a871f96107fa73af0594a1626527dc3abfe9