VLSIDA/OpenRAM #113: Potential bug in control logic write enable

lekez2005, 2021-02-27:
I suspect there may be a bug in the control logic definition for the write enable signal. The current logic [here](https://github.com/VLSIDA/OpenRAM/blob/master/compiler/modules/control_logic.py#L639) uses`self.connect_inst([input_name, "rbl_bl_delay_bar", "gated_clk_bar", "w_en", "vdd", "gnd"])`. Should it instead be `self.connect_inst([input_name, "rbl_bl_delay", "gated_clk_bar", "w_en", "vdd", "gnd"])`? i.e. **rbl_bl_delay_bar** to **rbl_bl_delay**? (Also a one-line fix for layout [here](https://github.com/VLSIDA/OpenRAM/blob/master/compiler/modules/control_logic.py#L655)) The logic as it is doesn't seem to work in simulation. Using `rbl_bl_delay_bar`, the write enable signal is only activated very briefly between falling edge of the `clk` and rising edge of `rbl_bl_delay` which is generally insufficient for writes.

[image]

vs 

[image]

mguthaus, 2021-02-27:
Hi,
Can you elaborate what configuration you are using? 
Thanks,
Matt

lekez2005, 2021-02-27:
It's a 64x64 freepdk45 configuration. I made some local changes to get extracted Calibre simulations working but those are unrelated to the control_logic issue highlighted above. Here is my full config file:

```
word_size = 64
num_words = 64

num_rw_ports = 1
num_r_ports = 0
num_w_ports = 0
num_banks = 1
words_per_row = 1
spice_name = "hspice"

tech_name = "freepdk45"
nominal_corner_only = False
process_corners = ["TT"]
supply_voltages = [1.0]
temperatures = [25]

route_supplies = True
perimeter_pins = False
check_lvsdrc = True
nominal_corner_only = True
load_scales = [0.5]
slew_scales = [0.5]
use_pex = True
analytical_delay = False

output_name = "sram_w_{0}_{1}_{2}".format(word_size, num_words, tech_name)
output_path = "macro/{}".format(output_name)
```

mguthaus, 2021-02-27:
Could you do a pull request on your local modifications?

Do you have stimulus to replicate this?

The rbl_delay signals are used in some of the control logic to address race
conditions. I need to look more closely at your simulation.

mguthaus, 2021-03-08:
I believe that the existing logic is correct. The write enable is "pulsed" for a duration of the delay line/replica line. This prevents a race condition at the next positive edge between the write enable signal and the precharge. If it is not properly being written, then it is possible that this pulse is not long enough and the delay line needs to be increased. Right now, that is not set adaptively, but should.

closed 2021-03-08 by mguthaus