VLSIDA/OpenRAM #115: regression test fail: subunit file missing

bvhoof, 2021-03-10:
Hi,

I pulled commit da3a1003019e31fb8cf8089bb568b093486a23c2
now, I cannot run the regression tests any more. it seems the subunit.py file is not included.

Best regards,
Bob.

bvhoof, 2021-03-10:
It seems to have come from commit 19e99d1.
Then, you added the parallel regression testing.

mguthaus, 2021-03-10:
It requires an additional Python dependency. All the dependencies are listed in a requirements.txt now as well so you can install with pip.

mguthaus, 2021-03-10:
Note, I need to update the README about these requirements.

mguthaus, 2021-03-11:
> Oh, I was not aware that this was a python package. I also needed to
> install scikit-learn.
> I run your tool in an anaconda environment. Do you want me to put an
> environment yaml in the main git repo? This might make things easier for
> other people to use.
>
> Best,
> Bob.

Sure, that would be great. I personally don't use anaconda but I know that
some may.

mguthaus, 2021-07-29:
I created a requirements.txt for the Python requirements. This can be used with pip install.
5a0c75d573dfa7cab187b86f910c26c4e2f59618

closed 2021-07-29 by mguthaus