VLSIDA/OpenRAM #118: Frequency for power numbers in report

oscargus, 2021-04-24:
It would make sense to write the assumed frequency next to the power numbers in the report. 

As far as I can tell, these are for 100 MHz as stated in the `tech.py` file using the `default_event_frequency` parameter.

I've tried to read-up about the liberty format and I have not found anything there (so far). Maybe it is obvious that the results are for 100 MHz because of the liberty format? However, in that case, why can it be modified in `tech.py`?

Either way, I believe it to be useful to state `@100MHz` or something similar in the report file (based on the used value). I can provide a PR if deemed beneficial.

mguthaus, 2021-04-26:
Hi Oscar, I believe that the power is in milliwatts so it depends on time already which will be taken care of by the power analysis tool. 

There is a lot to be improved in the power models, but we are focusing on verification and delay right now. If you have specific enhancements, please submit a PR to the dev branch and I will take a look. If you notice a bug, please also let us know.

closed 2021-04-26 by mguthaus

oscargus, 2021-04-26:
My point was that the dynamic power depends on clock frequency and that the
clock frequency used to get the power number is not given (but can be found
by reading the sources).

I'll try to submit a PR to show the consequences.

mguthaus, 2021-04-26:
I only give the .lib file which is read into another tool to do power
analysis like PrimeTime or OpenSTA or OpenTimer. You give those tools the
clock frequency in the setup.

Matt