VLSIDA/OpenRAM #120: Memory access energy

rubenheyrman, 2021-07-26:
Hi, 

Thank you for providing the tool!
I was wondering how you can extract the memory read and write access energy based on the datasheets generated (as in the attachment)?

Thanks in advance! 
Kind regards,

Ruben
[sram_128_216_freepdk45.zip](https://github.com/VLSIDA/OpenRAM/files/6876776/sram_128_216_freepdk45.zip)

mguthaus, 2021-07-26:
There is a section titled "Power Data" on the data sheet that has this information. Be aware that simulated power data seems to vary quite a bit depending on the simulator. In addition, the analytical power models are probably not that well calibrated. If you have any suggested fixes, please submit a PR to the dev branch.

rubenheyrman, 2021-07-26:
Thanks for your reply. If I assume the analytical results are correct,
which timing data do I have to multiply with the power to obtain the read
and write access energy? Or do you just divide by the clock frequency?

Kind regards,
Ruben