VLSIDA/OpenRAM #123: Can't find Sky130 bitcell

bkoppelmann, 2021-09-15:
Hey Matt,
I was trying out the Sky130 branch. When I was running OpenRAM with this config,
```
word_size = 2
num_words = 16
tech_name = "sky130"
nominal_corner_only = True
output_path = "temp"
output_name = "sram_{0}_{1}_{2}".format(word_size,num_words,tech_name)
```
I'll get the following error:
```
ERROR: file design.py: line 46: Custom cell pin names do not match spice file:
['BL', 'BR', 'VGND', 'VPWR', 'VPB', 'VNB', 'WL'] vs []
Traceback (most recent call last):
  File "OpenRAM/compiler/openram.py", line 54, in <module>
    c = sram_config(word_size=OPTS.word_size,
  File "OpenRAM/compiler/sram/sram_config.py", line 44, in __init__
    self.compute_sizes()
  File "OpenRAM/compiler/sram/sram_config.py", line 58, in compute_sizes
    bitcell = factory.create(module_type=OPTS.bitcell)
  File "OpenRAM/compiler/sram_factory.py", line 142, in create
    obj = mod(name=module_name, **kwargs)
  File "OpenRAM/technology/sky130/modules/sky130_bitcell.py", line 29, in __init__
    super().__init__(name, cell_name=cell_name, prop=props.bitcell_1port)
  File "OpenRAM/compiler/bitcells/bitcell_base.py", line 21, in __init__
    design.design.__init__(self, name, cell_name, prop)
  File "OpenRAM/compiler/base/design.py", line 46, in __init__
    debug.check(prop.port_names == self.pins,
  File "OpenRAM/compiler/debug.py", line 33, in check
    assert 0
```
This looks like that the custom bitcell is missing. I found in sky130_bitcell.py that it should be named `sky130_fd_bd_sram__sram_sp_cell_opt1` and looked for it in the skywater pdk. However I didn't find it there. Is this an oversight? I mean I can extract the cells form `https://github.com/efabless/sky130_sram_macros/tree/main/sky130_sram_1kbyte_1rw1r_8x1024_8`, but that seems cumbersome. Hope you can help.

Cheers,
Bastian

mguthaus, 2021-09-15:
I'm waiting on Google to push the cell repo into the public... This isn't merged into dev yet because it isn't functional yet.

closed 2021-09-15 by mguthaus

whutddk, 2022-09-07:
I met the same error when generating the smaller SRAM for MPW7, how can I make it?

mguthaus, 2022-09-08:
You need to install the library with "make install". Please make sure to use our docker environment to ensure that you have compatible versions.