VLSIDA/OpenRAM #126: Delay in generated SRAM

Jianfengusa, 2021-10-31:
Hi, I am new to SRAM design, and I am using a simple configuration to generate a 128*8 SRAM. Below is the configuration.

I have no issue using OpenRam, but I do have some questions about the generated files. 

In the generated Verilog file, there is a parameter "DELAY=3" with the comment "FIXME: This delay is arbitrary". 

I would like to know if this delay can be changed to 1, if so under which clock frequency and size of SRAM, and how would the change affect the PNR result from OpenRam. 

I really appreciate your help!! 

# Data word size
word_size = 8
# Number of words in the memory
num_words = 128 # 2^7
tech_name = "scn4m_subm"

# Port configuration (1-2 ports allowed)
num_rw_ports = 1 # single path for both read and write
num_r_ports = 0  # read only
num_w_ports = 0  # write only

# You can use the technology nominal corner only
nominal_corner_only = True

# Output directory for the results
output_path = "temp"
# Output file base name
output_name = "sram_{0}_{1}_{2}".format(word_size,num_words,tech_name)

mguthaus, 2021-10-31:
Hi,
This is a Verilog simulation delay only and has nothing to do with the actual delay of the SRAM which is available in the .lib file used by P&R. To get accurate delays, you would need to enable characterization but it is very slow. Our analytical models can provide an estimated delay but will be less accurate.

Jianfengusa, 2021-10-31:
Hi,

Thank you for the reply. 

Since the delay is only for Verilog simulation, then it can be set to 1 for simulation purposes, but this may not represent the correct behaviors of actual SRAM. If I want to simulate the actual behaviors of SRAM I will need to refer to the delay in the .lib file which can be obtained by enabling characterization in OpenRam. Am I understanding you correctly?

mguthaus, 2021-10-31:
Usually Verilog isn't used for accurate delay simulation. Instead, you should be relying on the static timing analysis. But, yes, you could add a more realistic delay from the .lib.

closed 2021-10-31 by mguthaus

mguthaus, 2025-03-16:
The delay is not the clock period.

mguthaus, 2025-07-12:
There is no pipelining in OpenRAM at this time. You will have to wait two cycles or come up with an alternative architectural solution.