VLSIDA/OpenRAM #127: sram attribute has no attribute 'control size'

MaestroLiu3, 2021-11-29:
[image]
I cannot seem to figure out the issue with the control size. I am currently trying to create 2 bank implementations but when I change the bank size to 2 and num_of_row = 1 to get the bank number I would get this error. The error is in compute_bus_sizes() module in the sram_base class in sram_base.py. I couldn't find self.control_size attribute either by going through the class hierarchy of the sram_base class, which itself inherits from desgin, verilog, and lef classes. self.control_size here I think is supposed to represent the number of control signals. Any help would be greatly appreciated.

mguthaus, 2021-11-29:
Hi,

The two bank option has fallen out of date and is not supported right now.
Your best bet is to create two SRAMs and combine their address space in
Verilog.

Sorry,
Matt

mguthaus, 2021-11-29:
Also, the row size is a derived parameter and is not meant to be set manually. Users should typically only set the word size and number of words. 

It seems that someone may have removed the code prohibiting 2 banks. We will need to add that back until we fix multi-bank.

mguthaus, 2022-07-21:
We have removed a hard 2 bank option in liu of a "soft" two bank option.

closed 2022-07-21 by mguthaus