VLSIDA/OpenRAM #128: How can I generate dual port SRAM

microSharjeel, 2021-12-07:
Hi all,
I want to generate a single dual port  SRAM. But I am getting error when I set parameters in configuration file as follows

num_rw_ports = 1
num_r_ports = 0
num_w_ports = 0

I get this error 
ERROR: file design.py: line 46: Custom cell pin names do not match spice file:
['bl0', 'br0', 'bl1', 'br1', 'wl0', 'wl1', 'vdd', 'gnd'] vs []
Kindly guide me in this regard,
best regards,
Sharjeel

mguthaus, 2021-12-07:
Your configuration is for a single port which doesn't match the error which is for dual port. Can you share your exact setup and command?

microSharjeel, 2021-12-07:
Really very sorry 
it was as follows
num_rw_ports = 2
configuration file is as follows

# Data word size
word_size = 32
num_rw_ports = 2
num_r_ports = 0
num_w_ports = 0

# Number of words in the memory
num_words = 1024

# Technology to use in $OPENRAM_TECH
tech_name = "sky130A"

# You can use the technology nominal corner only
# nominal_corner_only = True

process_corners = ["SS", "TT", "FF"]
# process_corners = ["TT"]

# Voltage corners to characterize
supply_voltages = [ 1.8 ]
# supply_voltages = [ 3.0, 3.3, 3.5 ]

# Temperature corners to characterize
# temperatures = [ 0, 25 100]

# Output directory for the results
output_path = "temp"
# Output file base name
output_name = "sram_{0}_{1}_{2}".format(word_size,num_words,tech_name)

# Disable analytical models for full characterization (WARNING: slow!)
# analytical_delay = False

microSharjeel, 2021-12-07:
really very sorry it was 
num_rw_ports = 2

microSharjeel, 2021-12-07:
configuration parameters from configuration file are as follows
# Data word size
word_size = 32
num_rw_ports = 2
num_r_ports = 0
num_w_ports = 0

# Number of words in the memory
num_words = 1024

# Technology to use in $OPENRAM_TECH
tech_name = "sky130A"

# You can use the technology nominal corner only
# nominal_corner_only = True

process_corners = ["SS", "TT", "FF"]
# process_corners = ["TT"]

# Voltage corners to characterize
supply_voltages = [ 1.8 ]
# supply_voltages = [ 3.0, 3.3, 3.5 ]

# Temperature corners to characterize
# temperatures = [ 0, 25 100]

# Output directory for the results
output_path = "temp"
# Output file base name
output_name = "sram_{0}_{1}_{2}".format(word_size,num_words,tech_name)

# Disable analytical models for full characterization (WARNING: slow!)
# analytical_delay = False

microSharjeel, 2021-12-07:
target from my makefile that calls OpenRAM to generate the RAM is as follows

sram: openram_config.py
	python3 $(OPENRAM_HOME)/openram.py openram_config.py
	mv temp sram

microSharjeel, 2021-12-07:
Sir 
I have found that this error is only with Skywater technology

mguthaus, 2021-12-07:
The technology should be "sky130" without the A. 

Make sure you also run "make install" in the OpenRAM directory to fetch the sky130 technology and library files.

microSharjeel, 2021-12-07:
I have cloned OpenRAM repo and when I do make install in the root directory I get this error
mkdir -p /../technology/sky130/gds_lib
cp: missing destination file operand after '/../technology/sky130/gds_lib'
Try 'cp --help' for more information.
make: *** [Makefile:75: /../technology/sky130/gds_lib] Error 1

mguthaus, 2021-12-07:
Change this line in the Makefile:
SRAM_GIT_REPO ?= git@github.com:google/skywater-pdk-libs-sky130_fd_bd_sram.git
to:
SRAM_GIT_REPO ?= https://github.com/google/skywater-pdk-libs-sky130_fd_bd_sram.git
I don't believe this has made it in yet.

You can ask further questions in #openram on the skywater slack...

mguthaus, 2021-12-07:
You can get this change (and some others) in the dev branch.

closed 2021-12-07 by mguthaus