VLSIDA/OpenRAM #129: "LVS mismatch" occurs when running example_configs

jingyao-zhang, 2022-01-13:
Hi,
Hope you all are doing well.

When running example configs, such as [example_config_1rw_1r_scn4m_subm.py](https://github.com/VLSIDA/OpenRAM/blob/stable/compiler/example_configs/example_config_1rw_1r_scn4m_subm.py) in the folder [example_configs](https://github.com/VLSIDA/OpenRAM/tree/stable/compiler/example_configs), "LVS mismatch" always occurs.
Does this matter?
I am using [Magic](http://opencircuitdesign.com/magic/) 8.3.254, and [Netgen](http://opencircuitdesign.com/netgen/) 1.5.217.

mguthaus, 2022-01-13:
Hi,

We are aware of some issues with the newer versions of magic that extract
extra ports. We will get back to you as soon as possible..

Matt

proppy, 2022-02-25:
This does sounds like what was discussed in https://github.com/The-OpenROAD-Project/OpenLane/pull/917#issuecomment-1037826636 and that @RTimothyEdwards fixed with https://github.com/RTimothyEdwards/magic/commit/2e99d0cff72fd7b82706e95ba0a8c01bbccda534.

mguthaus, 2022-02-25:
@proppy Thanks, but this is a bit different. It has to do with substrate port extraction. I think Tim submitted a fix yesterday but we are working on some other issues that cropped up do it now.
https://github.com/RTimothyEdwards/magic/commit/db4fa65bfc096e63954b37b188ea27b90ab31839

proppy, 2022-02-25:
@mguthaus Thanks for the pointer, I was able to reproduce the error in:
https://gist.github.com/proppy/1054e1618f5f90ce3af529ab5e56ad3f
```
ERROR: file magic.py: line 349: sky130_sram_1kbyte_1r1w_8x1024_8	LVS mismatch (results in /tmp/openram_root_2819_temp/sky130_sram_1kbyte_1r1w_8x1024_8.lvs.report)
```

Happy to try against `8.3.272` once it shows up in https://anaconda.org/LiteX-Hub/magic/files

mguthaus, 2026-10-16:
Closing as stale. No activity for about 4.6 years and no reproduction against current `dev`, so there is nothing actionable here as it stands.

Superseded in practice by #220 and #292, which cover sky130 LVS mismatches against current code and are both still open. If your mismatch was on a non-sky130 technology, that is worth a fresh issue, since freepdk45 and scn4m_subm LVS both pass in the current regression.

To be clear this is closed for inactivity, not rejected. If it still affects you please reopen or file fresh, with a reproduction against current `dev` including your configuration file, technology, OS and `python3 --version`, and the temp-directory contents OpenRAM leaves on failure ([the FAQ lists which files](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself a PR against `dev` would be welcome and we will review it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md).

closed 2026-10-16 by mguthaus