VLSIDA/OpenRAM #132: Can this complier generate arrays with customized SRAM cells?

AlwaysWry, 2022-02-11:
I'm a researcher on in-memory computing. Is this tool capable of generating SRAM arrays with customized cells (like the 8T or 10T cells)? It might be helpful for my project. Can I manage that by following the instructions in "porting to a new technology"?

mguthaus, 2022-02-11:
Hi,

It doesn't support this directly right now but there are a number of people working on it. The framework is useful for this. Let us know if you have any specific questions. Our Slack channel is probably the best place.

im-world, 2022-04-11:
Hello,

I would like to know how does one go about extending OpenRAM's functionality to a custom memory cell.

For the past few months, we have been steadily working towards understanding openRAM's code line-by-line, starting from openram.py, with the intent to extend openRAM to custom memory bit cells such as 8T and 10T (which would then be used for memories having in-memory computation).

We feel that our current approach to working on this extension is a bit cumbersome. Not only will understanding the entire codebase take time, but it might also be unnecessary for our purposes. Could you please provide us with a roadmap of how to go about this and the key OpenRAM modules we would need to understand?

&nbsp;
Thank you

&nbsp;
P.S. - It seems like the OpenRAM Slack channel is only open to persons having a '@ucsc.edu' email address. Is it possible for me and my collaborator to join the same (we have an '@iiti.ac.in' email address)?

AlwaysWry, 2022-04-11:
According to the [documentation](https://docs.google.com/presentation/d/10InGB33N51I6oBHnqpU7_w9DXlx-qe9zdrlco2Yc5co/edit#slide=id.g51f6eff2c1_2_935),  page 6 presents a 10T bitcell with isolated read port. So it seems feasible to generate 8T and 10T cells, maybe you can try by modifying the configuration file. However, I think the key question for CIM of this tool is the limited wordline signal design.

mguthaus, 2022-04-11:
There's an invitation link to the slack channel in the main README.

You can override any module, including bitcells,  in the configuration file, but there will be debugging needed. This is why we provide the array of unit tests that are mostly ordered bottom up. 

I'm not sure what the limitation on the word line signal is referencing...

im-world, 2022-04-13:
Hello Again

The invitation link to the slack channel present in the README redirects to the Workspace joining page, as expected. However, the joining page says that for joining the workspace, a **@ucsc.edu** email address is _necessary_ (and asks to contact the workspace administrator if you don't have a mail address on that domain). We are unable to join the slack channel for this reason.

When you say configuration files, do you mean the .py files located at `/compiler/bitcells/`, or are you referring to some other set of files?

It would be instructive if you could share an example repository having a non-6T bitcell (as there are people already working on it).

<br/>
Thank you