VLSIDA/OpenRAM #133: SRAM Macros for MPW Designs

mousaq92, 2022-03-30:
Hello,

I want to use sram_2kbyte_1rw_1r_32_512_8 for an MPW design. I cannot find the SS and FF .lib files for this macro in https://github.com/efabless/sky130_sram_macros

I am unable to generate this block in OpenRAM. Can I please get the files for this macro?

Thanks,

Mousa

mguthaus, 2022-04-02:
Hi,
We have not yet released valid lib files. The current one is also a placeholder with analytical estimates that aren't accurate...

mguthaus, 2026-10-13:
Closing as stale. This has been inactive for 4.5 years and there is no current reproduction against `dev`, so it is not actionable as it stands.

Pregenerated sky130 macros live in [VLSIDA/sky130_sram_macros](https://github.com/VLSIDA/sky130_sram_macros), which is usually what people want here.

To be clear, this is being closed for inactivity rather than rejected. If it still affects you, please reopen or file a fresh issue with a reproduction against current `dev`, including your configuration file, technology, OS and `python3 --version`, and the contents of the temp directory OpenRAM leaves behind on failure ([the FAQ explains which files those are](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself, a PR would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch.

closed 2026-10-13 by mguthaus