VLSIDA/OpenRAM #134: Clock frequency

cansudemirkiran, 2022-04-04:
Hi, 
I was wondering how we can relate to the maximum possible clock frequency/memory access latency we can operate the SRAM array using the timing values shown in the generated datasheet. I noticed that there is an event frequency value in the tech.py files which is set to 100 MHz. What does event frequency mean in this context? I tried changing the event frequency value and observed that the power numbers are scaled linearly but the timing values stayed the same. I'd greatly appreciate your explanation. Thanks!

mguthaus, 2022-04-04:
Hi Cansu,

The default is to use the analytical models (not characterization). This
will use Cacti-like models for delay and power. Therefore, power is just
alpha*CV^2f. I'm not sure what you mean by "event frequency" value...

cansudemirkiran, 2022-04-04:
Thank you for the quick response. Simply, I would like to obtain the information here in this [slide](https://docs.google.com/presentation/d/10InGB33N51I6oBHnqpU7_w9DXlx-qe9zdrlco2Yc5co/edit#slide=id.g188a3e06c5_1_62), the right hand side plots. 

Did you obtain these plots in the slides by using characterization? Do you have a documentation for the characterization option, how to use them instead of the default analytical models, etc. 

For your question, the event frequency is defined in the tech files ([example](https://github.com/VLSIDA/OpenRAM/blob/f66aac3264598eeae31225c62b6a4af52412d407/technology/freepdk45/tech/tech.py#L445))

Thanks!

mguthaus, 2022-04-11:
Hi,

We ran that manually using some bash scripts.

Matt

closed 2022-04-11 by mguthaus