VLSIDA/OpenRAM #135: Read Data Out Characteristics

stephenry, 2022-04-10:
Somewhat naive question, but my understanding of the read characteristics of the generated macros wasn't clear from the documentation. After a read has completed, and the data has become available at the output, if there is no subsequent operation (or the clock is gated), does the output remain valid indefinately?

mguthaus, 2022-04-10:
No, it will only remain valid for that read cycle. At the next clock, it will become invalid no matter whether the operation is a write, read or noop.

closed 2022-04-10 by mguthaus

stephenry, 2022-04-11:
Does this apply too if the clock is gated once the data is available?

mguthaus, 2022-04-11:
Yes, it does. The sense amps get shut off so the output will be invalid.