VLSIDA/OpenRAM #139: Execution performance of OpenRAM

FJShen, 2022-04-22:
Hi,

I was using OpenRAM for the first time, trying to generate an SRAM of 256kb size with FreePDK45 technology. It's already been running for 40 hours and still has not finished yet. Does it always take this long? Or am I doing something wrong? 

This long run time really surprises me because there is a tutorial [https://cornell-ece5745.github.io/ece5745-tut8-sram/](url) claiming it only takes a few minutes to generate a 4kb memory.

mguthaus, 2022-04-22:
It shouldn’t but there could be instances where the power routing is difficult or characterization takes a long time. 256kb is getting big too for a single array but it should be do-able. What settings and version are you using?

the hard multi banking doesn’t work right now so for real big memories you need to wrap them with verilog banks. We have a “soft banking” feature being tested now for this.

FJShen, 2022-04-22:
I'm using v1.1.19 (commit f66aac3). 
This is what my config looks like, it's intended for GPU register banks which have very wide ports. But looks like it is compiling for only one bank. 

> Technology: freepdk45                                                                                
> Total size: 262144 bits                                                                                                                                      
> Word size: 2048                                                                                                
> Words: 128                                                                                                       
> Banks: 1                                                                                                                                                                                              
> Write size: None                                                                                                            
> RW ports: 0                                                                                                                    
> R-only ports: 1                                                                                                                  
> W-only ports: 1

FJShen, 2022-04-22:
I'm using v1.1.19 (commit f66aac3). 
This is what my config looks like, looks like it is compiling for only one bank. 

> Technology: freepdk45
> Total size: 262144 [...]

mguthaus, 2022-04-22:
That’s an unusually large word size so I’ll have to check that out. It could be a runtime bug in our channel router.

FJShen, 2022-04-22:
Ok thanks. FYI here's what the tool output to the terminal, including 2 warning messages.  

>Technology: freepdk45
> Total size: 262144 bits
> Word size: 2048
> Words: 128
> Banks: 1
> Write size: None
> RW ports: 0
> R-only ports: 1
> W-only ports: 1
> DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
> DRC/LVS/PEX is disabled (check_lvsdrc=True to enable).
> Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
> Only generating nominal corner timing.
> Words per row: 1
> Output files are:
> ...
> ** Submodules: 643.4 seconds
> ** Placement: 61.2 seconds
> **** Retrieving pins: 0.3 seconds
> **** Analyzing pins: 0.2 seconds
> **** Finding blockages: 14374.0 seconds
> **** Converting blockages: 77.8 seconds
> WARNING: file pin_group.py: line 656:   Expanding conversion (addr1[5] layer=m3 ll=v[2583.52,-3.8975] ur=v[2583.655,-3.7625])
> WARNING: file pin_group.py: line 656:   Expanding conversion (addr1[6] layer=m3 ll=v[2583.52,-6.1075] ur=v[2583.655,-5.9725])
> **** Converting pins: 2417.9 seconds
> **** Separating adjacent pins: 60.0 seconds
> **** Enclosing pins: 3.0 seconds
> *** Finding pins and blockages: 20003.0 seconds
> [stuck here so far]

xobs, 2022-11-25:
Adding on to this, openram seems to take a very long time. `sky130_sram_1kbyte_1r1w_8x1024_8.py` has been running for an hour for me now. A process called `magicdnull` has 45 minutes of CPU time. Routing only took 815 seconds, but it's stopped there.

mguthaus, 2022-11-25:
This is extraction in magic. It is slow...

closed 2023-10-04 by mguthaus