VLSIDA/OpenRAM #140: SRAM macros for Open MPW

BYCakar, 2022-05-03:
Hi, I generated SRAM macros with OpenRAM, since I needed nearly 40 kB of RAM and it is almost impossible to provide it with 2 kB macros. I got Magic DRC errors and asked about it in the sky130 Slack channel. They told me that SRAM macros commonly have this issues and reccomended to turn off Magic DRC check while running user_project_wrapper. The flow passes but I have problems in precheck due to DRC issues. I also asked about it in the Slack channel and they told me that SRAM macros in the PDK directory are automatically excluded from DRC check, but they were not sure if users can add custom SRAM macros. Am I allowed to add my custom SRAM in caravel user project? If yes, how can I make the precheck tool to exclude my SRAM too?

mguthaus, 2022-05-03:
Hi,

The DRC issues in precheck are on the eFabless end and you should talk to them. We generate the macros, but need their help on integration in the flow.

In terms of the very large memories, we are still verifying them. I can't guarantee anything works, but we are confident in the 1kbyte and possibly 2kbyte and 4kbyte.

closed 2022-05-03 by mguthaus

BYCakar, 2022-05-04:
I also opened the same issue in caravel_user_project repository, but I haven't got any answers yet. Just in case I am able to implement generated SRAMs in user project area, how can I verify my SRAM?

BYCakar, 2022-05-26:
As far as I know, DRC issues with SRAM macros are common. So you should turn off Magic DRC when running user project wrapper. In the precheck stage checker excludes SRAM macros to ignore these DRC issues, thus your design will pass successfully. But I must indicate that I know this from slack discussions and my examinations on precheck scripts, I have not personally tried it yet.