VLSIDA/OpenRAM #141: LVS Errors with Sky130 SRAMs

BYCakar, 2022-05-05:
Hi, I tried to verify some SRAM macros, but I get an LVS error. I examined what the problem comes from and I think the reason is that one of the files are missing. In sky130_sram_lvsdrc_test.lvs.out file, I get these messages:
```
Reading netlist file sky130_sram_lvsdrc_test.spice
Reading netlist file sky130_sram_lvsdrc_test.sp
...
Contents of circuit 1:  Circuit: 'sky130_sram_lvsdrc_test'
Circuit sky130_sram_lvsdrc_test contains 0 device instances.
Circuit contains 0 nets.
Contents of circuit 2:  Circuit: 'sky130_sram_lvsdrc_test'
Circuit sky130_sram_lvsdrc_test contains 4346 device instances.
  Class: sky130_fd_pr__nfet_01v8 instances: 742
  Class: sky130_fd_pr__special_pfet_pass instances: 648
  Class: sky130_fd_pr__special_nfet_latch instances: 2160
  Class: sky130_fd_pr__pfet_01v8 instances: 796
Circuit contains 1189 nets.
```
It tries to read sky130_sram_lvsdrc_test.spice, but there is no such file.

I added all the output files in the tarball, including lvs outputs and configuration file, so you can reproduce them.
```
[openram_logs_and_config.tar.gz](https://github.com/VLSIDA/OpenRAM/files/8630388/openram_logs_and_config.tar.gz)
```

mguthaus, 2022-05-06:
This is very sensitive to the version of Magic and Netgen. There were some bugs with substrate extraction that have evolved over time. I'm going to defer on this until I push the new version any day now which will have a docker image with versions that work.

BYCakar, 2022-05-09:
I need to verify my designs, how can I deal with this? Working magic and netgen versions could be helpful maybe. I am using OpenRAM with version v1.1.19.

mguthaus, 2022-05-24:
I pushed the versions in our Dockerfile that can solve this issue. They are in the dev branch now.

closed 2022-05-24 by mguthaus