VLSIDA/OpenRAM #144: SRAM Simulation Codes

BYCakar, 2022-05-31:
Hi, I was simulating my design and I realized there might be a mismatch between verilog simulation code and the description in OpenRAM paper.

[image]
According to this figure, read and write data are available one cycle after the operation, like most of the synchronous RAMs do.

[image]
In verilog simulation, read data is available two cycles after the read. I think the last always block below causes this in verilog code.

```
  // All inputs are registers
  always @(posedge clk0)
  begin
    csb0_reg = csb0;
    web0_reg = web0;
    wmask0_reg = wmask0;
    addr0_reg = addr0;
    din0_reg = din0;
    #(T_HOLD) dout0 = 32'bx;
    if ( !csb0_reg && web0_reg && VERBOSE ) 
      $display($time," Reading %m addr0=%b dout0=%b",addr0_reg,mem[addr0_reg]);
    if ( !csb0_reg && !web0_reg && VERBOSE )
      $display($time," Writing %m addr0=%b din0=%b wmask0=%b",addr0_reg,din0_reg,wmask0_reg);
  end

  // Memory Write Block Port 0
  // Write Operation : When web0 = 0, csb0 = 0
  always @ (negedge clk0)
  begin : MEM_WRITE0
    if ( !csb0_reg && !web0_reg ) begin
        if (wmask0_reg[0])
                mem[addr0_reg][7:0] = din0_reg[7:0];
        if (wmask0_reg[1])
                mem[addr0_reg][15:8] = din0_reg[15:8];
        if (wmask0_reg[2])
                mem[addr0_reg][23:16] = din0_reg[23:16];
        if (wmask0_reg[3])
                mem[addr0_reg][31:24] = din0_reg[31:24];
    end
  end

  // Memory Read Block Port 0
  // Read Operation : When web0 = 1, csb0 = 0
  always @ (negedge clk0)
  begin : MEM_READ0
    if (!csb0_reg && web0_reg)
       dout0 <= #(DELAY) mem[addr0_reg]; // THE ISSUE IS HERE!!!
  end
```

I think unblocking assignment causes the simulation to wait an additional cycle which causes a mismatch. I am probably missing something, maybe not driving some signals properly, but I need to use SRAM macros like described in OpenRAM paper. How can I fix this?

mguthaus, 2022-05-31:
The paper is outdated and we don't have a way to update it since it was published. Please look at the documentation presentation in the README.md which should be more up to date.

The simulation model is accurate to the expected behavior.

mguthaus, 2022-05-31:
Also, it is available at the end of a cycle, not "two cycles"... It uses the second half for the operation, so it requires two *edges* but not cycles.

closed 2022-05-31 by mguthaus

BYCakar, 2022-05-31:
In the documentation, at the first positive edge csb is low, at the next positive edge the data is fetched. It really requires two edges. But in the simulation, at the first positive edge csb is low, at the next positive edge there is no data, the data is fetched a positive edge later. Thus I said "two cycles". So I still think something is wrong with my simulation.

mguthaus, 2022-06-01:
You need the csb, web, address, and data asserted before the positive edge.
In your simulation it is at the positive edge which will result in a
Verilog race condition. In this case, it doesn't receive it before the
edge. Hope that helps.

Matt