VLSIDA/OpenRAM #146: OpenRAM overflow error

harishMadhavan1010, 2022-06-18:
[image]

[image]

I am facing overflow error whenever I simulate the above file and for some reason, the error does not show up if I keep the word_size and num_words less (@2,16). I think that this is a bug. I'd be glad if someone helps me out. Meanwhile, I'll mess around with different word sizes, number of words and corners. Thank you!

mguthaus, 2022-06-25:
Which commit of OpenRAM are you using?

mguthaus, 2022-06-25:
The technology should be sky130 not sky130A in your config.

Ryabkomv, 2022-07-21:
I have the same issue. Setting the tech to sky130 in config file throws SystemError: Unable to find open_pdk tech file. Set PDK_ROOT.

mguthaus, 2022-07-21:
Yes, you need to set the PDK_ROOT so that it knows where to install the PDK when you run:

make pdk
make install

open_pdks, specifically was added in c8139cf1452c13f98e200e24a45f1c79d91c9c27. 
This will assume that this is the local OpenRAM directory and just put it there if you don't define it. If you have the PDK installed for OpenLane, it *may* work but that might also be a different version which could cause problems so it is recommended that you install using our Makefiles.

closed 2022-07-21 by mguthaus