VLSIDA/OpenRAM #148: Does OpenRAM support bit enable ?  How to configure ?

szuhzw, 2022-07-19:


mguthaus, 2022-07-21:
Please see the following example that uses byte write. You can change write_size to 1 to get byte write:
https://github.com/VLSIDA/OpenRAM/blob/stable/macros/configs/sky130_sram_2kbyte_1rw1r_32x512_8.py

closed 2022-07-21 by mguthaus

szuhzw, 2022-07-22:
I tried to write write_size to 1, but it would report error when runing.

szuhzw, 2022-07-22:
It's ok to write write_size to 8.

mguthaus, 2022-07-22:
If the and gate that does the masking is narrower than the bitcell in a given technology, it can be bitwise masked. This means it isn't so you would need to make a custom and gate to do it.

mguthaus, 2022-07-22:
The error would say this...

szuhzw, 2022-07-22:
So, you means the OpenRAM cannot support bite write at present?

mguthaus, 2022-07-22:
It supports it but not in that technology. It depends on the width of the and gate compared to the width of the bitcell which is different in each technology. You could add an improved and cell to fix this.

szuhzw, 2022-07-22:
OK,Thank you